pbatard/rufus #2442: Thanks

Flavio93Zena, 2024-03-10:
Used Rufus to make a bootable Ubuntu as I had previously done last year, it failed and locked my drive into read-only, completely impossible to fix. Screw this program, and this issue to close is just a very minor nuisance compared to the massive annoyance of having to throw away my beloved usb drive due to your program. Another "coincidence" I'm sure, it locked the drive to start writing on it, failed and never worked again.
So yeah, thanks.
Pissed system engineer.

pbatard, 2024-03-10:
And thank you for reading these two very relevant entries from the FAQ:
* https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive
* https://github.com/pbatard/rufus/wiki/FAQ#user-content-Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media

I'm always amazed at people coming here, at about the frequency you'd expect them to report **coincidental** drive failures for an application that gets downloaded close to 3 million times each month, and who always seem to have a strong willingness to berate the developer of the application, and yet don't seem to have the same dedication, if they **truly** are convinced that it's the software that caused their problem, to contact the manufacturer of their flash drive so that they investigate the issue to __actually validate their claim__, as it is obviously in the best interest of manufacturers, if software is allegedly able to damage flash drive, to __expose__ that software as a danger to users.

It's almost as if pissed off users just want to vent their frustration by blaming the software and its developer for their woes, while at the same time not being genuinely convinced that it's the software that caused their drive to fail....

closed 2024-03-10 by pbatard

Flavio93Zena, 2024-03-10:
I've read those, that's why there's the "coincidence" bit, should have been a strong hint at that.

pbatard, 2024-03-10:
A strong hint that you haven't properly read those is that you still went ahead and discarded everything that was explained in these entries to go on your __pure unfounded belief__ that there can be no such thing as a coincidence but, instead, that the finger must be firmly pointed at something or someone.

Flavio93Zena, 2024-03-10:
Much like many others before me have pointed out - I used your software, tried to do something I had previously done with no issues whatsoever, it bricked my drive and you will just brush it off as another naysayer/coincidence.
So at the end of the day you don't care but I'm out a drive I cared about, so yes, finger IS firmly pointed at this garbage of a software.

pbatard, 2024-03-10:
Put your belief where your mouth is then, and contact the manufacturer of your drive to prove that Rufus is the cause then.

Or does _"innocent until proven guilty"_ take a backseat in the place you are from because of your __assertion__ that there cannot be coincidental flash drive failures when using a flash drive and that electronics device do last forever?

Flavio93Zena, 2024-03-10:
Given it's out of warranty it's not a viable option but I can surely ask, they'll most likely tell me to contact you (lol).
I am totally fine with drive failures, it just eeeeever so happens to be the *exact* moment that it failed to flash the bootable Ubuntu iso on it eh. Had I been the only one I would have said sod it and whatever, but since I am not, no.

pbatard, 2024-03-10:
Then please do yourself a favour and never google _"balenaEtcher bricked my flash drive"_ or _"Ventoy bricked my flash drive"_ so you can keep leaving in your beloved finger pointing fantasy land where having a handful of people on the internet reporting that some software bricked their drive must mean that it's true.

Now, if you'll excuse me, I have to get into a meeting with the Etcher and Ventoy people where we will lay out new creative way to brick flash drives (because, since you flat out reject the possibility of a coincidence, the only explanation that works in your world is that we are all colluding together to damage user hardware).

Flavio93Zena, 2024-03-10:
It totally wasn't necessarily true until I witnessed it myself, and I never had anything like this happen before. Failed drives, usb drives yeah got all of those, but bricked into read-only and not even cmd or linux would get rid of it? No, that's a first, especially when it's not even fully broken - you can benchmark read speed off it but not write speed for obvious reasons, so it really is software-bricked, I checked it periodically and didn't have bad blocks or anything like that.

pbatard, 2024-03-10:
_"My car was running fine until it broke down. I've always been taking good care of it so the fault has to be with the road and not anything to do with the car itself."_

Flavio93Zena, 2024-03-10:
If you hit a pothole and it smashes your suspensions, yes. Smartass.

pbatard, 2024-03-10:
You missed the part where there's about 3 millions of other road users who went on the same road at around the same time as you and never saw any pothole, let alone hit one. Or the part where suspension manufacturers, who have a vested interest in exposing an alleged known pothole (people swear they have seen it!!) that is causing damage to suspensions and making them lose money on warranty, and who should be very determined to produce a picture of said pothole to alert people, somehow never got around to do just that.

Flavio93Zena, 2024-03-10:
A little bit harder to take a picture of something you can't see against a physical pothole, don't you think?

pbatard, 2024-03-10:
If something like a pothole does exist, it can be photographed. And what about all the other users who never hit that pothole of yours?

https://tooomm.github.io/github-release-stats/?username=pbatard&repository=rufus

Now why on earth is this public issue tracker not full to the brim of "Rufus damaged my flash drive!" but instead seems to only get the occasional report which you would associate with exactly the rate of coincidental failures against the download rate?

Where is your explanation for that? Somehow people become shy when they are pissed off at software? Or somehow I delete these reports (but then there should be gaps in issue number and other telltale signs)?

Flavio93Zena, 2024-03-10:
Because as I mentioned before I had used the program multiples times before and never had a problem, so the vast majority of times it will work correctly, except when it doesn't and bricks the device. Extremely rare but it's a damn thing or I wouldn't be here.
Also don't trust statistics, especially on things that are downloaded plenty of times it can be inflated a LOT, heck most likely at least 10 of those overalls are mine, and that's one person.

pbatard, 2024-03-10:
I like your logic where you happily divide downloads, but do not consider the possibility of multiplying the usage one person may make of the application (therefore multiplying the possibility of experiencing failures).

At this stage, it is clear that no factual argument will be able to convince you that you did experience a coincidental failure, because you somehow just happen to __know__ the truth of the matter (and will as confidently reject the counter arguments that are laid out in great details in the fact. Since you're a systems engineer, please look into Windows Hardware Abstraction Layer sometime, and how software applications really access USB flash drives, if you are that convinced that software can damage hardware such as USB flash drives) and will continue to twist whatever fact is provided to you to serve your narrative. 

So, since continuing to try to present facts to you is going to be pointless, I'm just going to lock this thread.