pbatard/rufus #2444: I ran (and cancelled) a run to check bad blocks and now my flashdrive is on "read-only". I only checked for bad blocks once.

Martin-L-H, 2024-03-18:
Yes, i did read the FAQ. Im aware there's a high chance my drive is totally bricked, however there's one thing i'd like to point out.

I BARELY used this drive, and since i bought it and took it from the box its been about 1 year and not really that much use. I only got to run a bad block check once (the most passes mode, the 4 one, but i cancelled before the first one even got to 1% due to the amount of time it was taking). Now the drive is on read-only, and even using window's CMD diskpart im unable to remove the readonly flag.

I have tried other tools but they all refuse to work due to the fact the drive is set on read only. Is there anything i can do? Its a Kingston Datatraveler Exodia 32gb. If there's a chance that a firmware flash could give life back to this thing im willing to do it. Im really more concerned about trying to fix this thing than in getting a new one or recovering any info from it.

Is it possible that Rufus does more damage by excessive overwrite than previously expected and wears the device down? Or is this a software issue that can be fixed and prevented in the future?

[image]

pineapple63, 2024-03-18:
From what I understand, the controller in a USB drive will switch the drive to read only mode if it detects something it isn’t happy with
It is technically possible to reset this read only flag, but i would not recommend it
And i would also not recommend trying to reflash the controllers firmware, as doing so could remove factory calibration which could make the drive even less reliable

pbatard, 2024-03-19:
> Is it possible that Rufus does more damage by excessive overwrite than previously expected and wears the device down?

There's no such thing as "excessive overwrite". Either a block of data gets written down or it doesn't get written. And there is very little metadata used by FAT32 or NTFS, which means that there aren't that many extra blocks written compared to the number of blocks that are needed to fit the sum of the bytes that is occupied by all the files from the ISO. Plus there is Windows caching which means that Rufus doesn't even write directly to the flash drive but instead it first (transparently) writes to RAM where Windows reorganises the data that needs to be written in large contiguous sequential sequences.

No matter how much some people would like to point the finger at Rufus (or other software, because if you look for "balenaEtcher bricked my drive" or "Ventoy bricked my drive" you will find about as many people who are adamant that it must be the software's fault) what you experienced was a **coincidental** failure.

And for what is worth, the last flash drive failure I experienced (that wasn't due to a fake drive - I had the unfortunate displeasure of being sold a fake SD card about 1 year ago on aliexpress from a seller that said they were official for the brand but where removed from the store before I could ask for a refund) was with a new drive, that I used about 3-4 times (and I don't think I even used Rufus on that drive) and left on a shelf for about a month. before trying to use it again only to find out it had become completely unreadable for absolutely no reason, and certainly not from any usage, as it was perfectly readable when I had used it last. That was about 5 years ago, and, despite using Rufus practically day in and day out on a variety of flash drives, I am fortunate to not have experienced any failure since, even though, if Rufus was really placing some kind of load on flash memory, I would statistically be prone to have seen some of the failure that people **believe** Rufus may incur.

So, really, all I can point to, before I close this issue, is [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media), and advise you to RMA your media since it should be fairly new and under warranty from Kingston.

closed 2024-03-19 by pbatard

Martin-L-H, 2024-03-19:
> From what I understand, the controller in a USB drive will switch the drive to read only mode if it detects something it isn’t happy with It is technically possible to reset this read only flag, but [...]

Do you know of any way to remove the read only mode then? Considering the little use of the device i have no idea what it could have triggered it, and i have no idea how to remove it since diskpart says "all good bro!" but it always keeps the read only mode. I tried gparted on linux to the same issue, it can do nothing.

I did try to ask Kingston about the warranty, but they say that its not covered due to misuse (misuse?) and then pointed me to an old HP storage formater that works just the same as the windows integrated one.

> There's no such thing as "excessive overwrite". Either a block of data gets written down or it doesn't get written. And there is very little metadata used by FAT32 or NTFS, which means that there [...]

That's the issue. I DONT think Rufus broke it. But the only explanation is that either a first party Kingston bought came a lemon or that in the middle of the bad sector check i messed something up software-side. If you believe that im being an idiot in asking for help through here or that i dont deserve help for this problem i wont contest it, but frankly im ripping my hair out trying to find a solution to make the drive usable and hopefully leaving for the record how to fix this issue if it happens to someone else again. I tried using the regedit tips, but there was no StorageDevicePolicies key there. Just in case i made one, i followed the instructions, restarted the computer, but no dice. I then tried again deleting the key, restarting computer, but no dice.

When i try using Rufus on the drive, it gets stuck in the "Clearing MBR/PBR/GPT structure" and throws this error:

[image]

I CAN believe that the drive is 100% totally royally utterly screwed. I just want to make sure that it is since its got less than 50 rewrites in it (i've always used it as installation media for different OS's) and less than 1 year since i bought it. I also have no idea if "Current Read-only state" is different from "Read-only" in the attributes.

I hope this ChipEasy report (or surprisingly lack of) can be a hint to know if there's a hardware problem going on:

[image]

pbatard, 2024-03-19:
Your drive is screwed. A healthy/working drive should **never** turn read-only. You are wasting your time trying to find a solution that could remove the read-only attribute, which only happened because the drive controller detected that your flash memory had failed. Replace your drive. The only proper solution is to replace your drive.

Martin-L-H, 2024-03-19:
> Your drive is screwed. A healthy/working drive should **never** turn read-only. You are wasting your time trying to find a solution that could remove the read-only attribute, which only happened [...]

Guess it was a lemon after all then.