pbatard/rufus #2445: [Help] Sandisk USB Stuck in DD Mode

TerraPlayz, 2024-03-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x I also tried one or more of the following:
  - [x] Using a different USB drive. - **N/A**
  - [x] Plugging the USB into a different port.
  - [*] Running Rufus on a different computer.
- [*] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have used rufus to burn Ubuntu 20.04 ISO in **DD Mode** to my usb. i would like to recover it.

I have tried the "Non Bootable" Method of FAQ and also tried diskpart to clear read only attributes but none of it worked.

Log
---
```
Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.3930)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {51A451FB-432E-4255-A9D4-BC668442E5FC}
Max parts: 128, Start Offset: 17408, Usable = 4004477440 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {7A1F1E04-4052-4DBE-A2E3-F3C4500E9552}
  Size: 3.7 GB (4003446272 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Could not get Disk Extents: [0x00000001] Incorrect function.

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Warning: Could not delete partition(s): [0x8007045D] The request could not be performed because of an I/O device error.
Could not get Disk Extents: [0x00000001] Incorrect function.
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
Could not get Disk Extents: [0x00000001] Incorrect function.
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Retrying in 5 seconds...
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Retrying in 5 seconds...
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Retrying in 5 seconds...
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Could not reset partitions
Re-mounted volume as F: after error

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {51A451FB-432E-4255-A9D4-BC668442E5FC}
Max parts: 128, Start Offset: 17408, Usable = 4004477440 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {7A1F1E04-4052-4DBE-A2E3-F3C4500E9552}
  Size: 3.7 GB (4003446272 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {51A451FB-432E-4255-A9D4-BC668442E5FC}
Max parts: 128, Start Offset: 17408, Usable = 4004477440 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {7A1F1E04-4052-4DBE-A2E3-F3C4500E9552}
  Size: 3.7 GB (4003446272 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Checking for Rufus updates...
Checking release channel...
No new release version found.
```

pbatard, 2024-03-21:
If you get `Write error [0x0000045D] The request could not be performed because of an I/O device error.` then your device is most likely dead and you will have to replace it.

Unfortunately, flash memory, and especially the one used in consumer drives, is not everlasting, and will fail as you use it (and using a brand named drive like SanDisk, while it should reduce the probability of a failure somewhat is no guarantee that it will not fail. As a matter of fact, the last flash drive failure I experienced, about 5 years ago, was with a SanDisk drive that I had barely used and that had been sitting on a shelf for about a month before I tried to use it again).

If I recall correctly, SanDisk drives usually have good warranties (at least I was able to get mine replaced without much of an issue), so I would encourage you to look into getting your drive RMA'd.

TerraPlayz, 2024-03-21:
> If you get `Write error [0x0000045D] The request could not be performed because of an I/O device error.` then your device is most likely dead and you will have to replace it.
>
> Unfortunately, flash [...]

my drive is out of warranty. but its pretty much not dead since i can still run the Ubuntu OS from it that i burned there. there i can still read/write in limited storage i have.

pbatard, 2024-03-25:
Some flash drive controllers turn drives to read-only when they detect an error, to give you some chance to read your data back. That still does not mean that your drive is not defective. On the contrary, that's a pretty good indication that, as far as the manufacturer of your hardware is concerned, it considers that your drive is not salvageable.

closed 2024-03-25 by pbatard

Omxpro, 2024-03-25:
> Some flash drive controllers turn drives to read-only when they detect an error, to give you some chance to read your data back. That still does not mean that your drive is not defective. On the [...]

Alright thanks 😊