pbatard/rufus #2446: ﻿Feature Request: Option to increase default size of generated windows recovery partition. (if possible)

Aerocatia, 2024-03-22:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Some Windows install media (like Windows 10 ISOs) will create too small of a recovery partition by default. This results in the failure to install [KB5034441](https://support.microsoft.com/en-gb/topic/kb5034441-windows-recovery-environment-update-for-windows-10-version-21h2-and-22h2-january-9-2024-62c04204-aaa5-4fee-a02a-2fdea17075a8) after a fresh install. The best thing Microsoft could come up with was tell users to resize the partitions. See: [KB5028997](https://support.microsoft.com/en-us/topic/kb5028997-instructions-to-manually-resize-your-partition-to-install-the-winre-update-400faa27-9343-461c-ada9-24c8229763bf?trk=article-ssr-frontend-pulse_little-text-block)

Doing this directly after a fresh install is very annoying.

I have no idea if Rufus can do anything about it, but since it can apply other install tweaks, is there any possible way to change the default size of the recovery partition by patching the install media on boot USB creation time?

If this is possible then having this feature would save a lot of trouble.

Log
---
```
<FULL LOG>
```

ONE11111111ONE, 2024-03-22:
i know it is annoying, but here is the solution at the same time (works for all fresh installed Win 10 isos)

CMD (Admin):

reagentc /disable
dir /a /s c:\winre.wim
reagentc /setreimage /path C:\$WinREAgent\Backup
reagentc /enable

Done. Close and Reinstall Update. Works fine.

pbatard, 2024-03-25:
Well, since this appears to be Windows 10 specific, and we're about 18 months from Windows 10 EOL, I don't see myself adding some Windows 10 specific quirks in Rufus.

As such, I'm afraid I will dismiss this request.

closed 2024-03-25 by pbatard

ONE11111111ONE, 2024-03-25:
@pbatard Sorry Sir, but do you know how many people (have already) move(d) to Win 10 Enterprise 2021 LTSC IoT? 
You will not be able to count. Since it's supported till 2032...there is a lot of space. And it affects everyone no matter which Win 10 version who makes a fresh clean install. Nobody wants this AI Win 11 or whatever comes total control OS.

Aerocatia, 2024-03-25:
Not to mention processors <Coffee Lake are stuck on Windows 10 without using the hardware bypass, which may or may not lead to sudden breakage due to software assuming available CPU features based on officially supported minimum hardware.

Anyway, for anyone who ends up here in the future, it *is* possible to customize this in `unattend.xml`, as OEMs use this system to customize what they have on the recovery image, however the documentation is rather obtuse so I don't have the lines you would need to paste in on hand.

Thank you for your time.

boktai1000, 2024-05-14:
I've been fighting this issue with installs for months, and been hoping that each new Rufus release (or an updated Microsoft ISO) would solve this issue, but unfortunately it has not.

My apologies on commenting on a closed thread, but I didn't want to open a new duplicate request and must agree with the comments above - it would be unfortunate to leave Windows 10 in this state, especially for those without the ability to upgrade.

pineapple63, 2024-05-14:
From what I know, even if they did want to look into adding in an option to have an increased recovery partition size, the only way i know of accomplishing that would be to configure the installer to automatically wipe the drive and create all of the Windows installation partitions, which it will do without prompting the user at all, which could be a problem

ONE11111111ONE, 2024-05-14:
Why not just using the solution mentioned above? No headaches..

pineapple63, 2024-05-14:
I should point out that with what i mentioned above, it can be accomplished by creating an autounattend file with the required config info and placing it in the root directory of a USB drive (it can be on the same one as the installer, or a seperate USB drive, and the file will be detected automatically by the installer)
Athough, be aware that doing this may conflict with any customisations selected in Rufus

ONE11111111ONE, 2024-05-15:
Thank you could be so easy. Yet m$ made this update in january and they offer a crazy solution on their website but no patch ever....this is in german forums the best way, and it works no matter how small your RE is. Don t you think m$ is well aware of what they did? Better don t go to derp trance hypnosis sleep... all i can say

thecatontheceiling, 2024-05-15:
since people here apparently have access to LTSC ISO images, just wanted to say you can get updated windows 10/11 consumer isos from the same place you **legally** got LTSC from. they do not have this issue and you do not have to wait for windows update after clean install. 

here's the filenames for the ISO's that MVS offers me for reference:
en-us_windows_11_consumer_editions_version_23h2_updated_**april_2024**_x64_dvd_d986680b.iso

en-us_windows_10_consumer_editions_version_22h2_updated_**april_2024**_x64_dvd_9a92dc89.iso

thecatontheceiling, 2024-05-15:
> @pbatard Sorry Sir, but do you know how many people (already have) move(d) to Win 10 Enterprise 2021 LTSC IoT? 
> You will not be able to count. Since it's supported till 2032...there is a lot of space. And it affects everyone no matter which Win 10 version who makes a fresh clean install. Nobody wants this AI Win 11 or whatever comes total control OS.

lmao, 99% of people using LTSC are not using it legally anyways. pbatard has no interest in supporting this one specific edition that needs a 6000$ a year subscription to access. he would much rather work on more important features than fix a minor inconvenience for a few people who **actually** have legal access to LTSC.

ONE11111111ONE, 2024-05-16:
i can write funny things now, since you are out of ideas to explain it in a positive manner.
you don't know what is going on with microsoft. At all. Period.

pbatard, 2024-05-16:
Well, if there are updated Windows 10 ISOs, as @thecatontheceiling pointed out above, I will stand by my choice not to do anything about this in Rufus. This is 100% a Microsoft created problem, that Microsoft needed to fix, and that they have now done. And I agree that, if you are planning to use Windows 10 past the general public EOL, then you should have no problem accessing these corrected ISOs.

If that is still not acceptable to you, then I will ask you to please redirect your energy towards Microsoft support and ask them to make the fixed ISOs available to the general public through https://www.microsoft.com/en-us/software-download/windows10ISO. If there is enough demand, I don't see why they shouldn't be able to do it, and it wouldn't be the first time Microsoft, who is still a consumer driven company, would have done something because consumers asked them to.

pineapple63, 2024-05-16:
With the thing i was saying about getting the installer to create the partitions automatically, if you want to set that up yourself, just look up instructions on how to do an unattended installation of windows
I could create a file and upload it (and configure it so it won’t prompt you to connect to WiFi during setup or sign into a microsoft account)

ONE11111111ONE, 2024-05-17:
@pineapple63 ok i will check it on the weekend and if i cannot do it i will come back to your help, cheers and thanks man.