pbatard/rufus #2447: Windows To Go User Account Control and System App Issues

NoobNotFound, 2024-03-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First of all I have to say
### Avoid paying too much attention on this issue.
**I'm not sure whether this is something wrong with me or Rufus but I need to have advice/help.**

I recently accidentally corrupted my main SSD, so I used rufus (using another PC) to create a Windows To Go for temporal use.
I think it's a decent USB (3.1 128GB) for this kind of thing I even hope to play Games on it (Games are on my HDD).  
But that's not the problem I want to solve, the problem is there is something wrong with the User Account Control and some other things. These are the problems that occurred

- The Management console does not work properly when using Win + X (it says device administrator blocked you from using this app. But it works when I used elevated CMD)
- The windows defender/Security works but the App doesn't open (It says this app can't open, I also tried to fix it using some ways)
- Most of the times like changing a setting, running task manager it asks to run as admin even though it is used to be run Without asking for user

I don't want to reinstall windows to my USB because I have installed, download some large files, apps to the USB
Although I'm not sure this is because I'm in a old **Insider Build** which has got **expired**  
I have additionally attached some screenshots for more details.

Build details
```
Edition	Windows 11 Home Insider Preview
Version	22H2
Installed on	‎3/‎22/‎2024
OS build	23466.1001
Experience	Windows Feature Experience Pack 1000.23466.1001.0
```
Device details
```
Processor	11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
Installed RAM	16.0 GB (15.7 GB usable)
System type	64-bit operating system, x64-based processor
```
The rufus I used - `rufus-4.4p.exe`  

[image]
[image]
[image]

Log
---
```
<FULL LOG>
```

NoobNotFound, 2024-03-24:
I should update to the latest version and check whether they still exist.

pbatard, 2024-03-25:
I'm not going to look into this, but I'm going to leave this issue opened for about a week, and then close it if nobody bothered to reply, because I really don't think you are posting in the best place to get an answer to your question.

closed 2024-03-31 by pbatard