pbatard/rufus #2448: Process/thread sometimes never exits even when forcibly terminated

arukateru, 2024-03-24:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
After terminating Rufus through a task manager, the process does not terminate and gets stuck at exiting in two separate occasions. One of the times, even 15 minutes later, this process and its one remaining thread was still "exiting". 

First time what I did:
Simply opened Rufus as normal and began to burn an image. Part way through, the application became stuck (not frozen, the timer was still going up) but I know that it was not still copying any file. There was no file in the image that would take that long to copy, and I double checked using Process Hacker, for some reason, the program was READING around 200 kb of IO for many minutes. So, I decided to cancel it. Same thing, process wasn't actually doing anything.

After waiting a long time for it, I finally got fed up and decided to end the process forcibly. However, many minutes later, it was still there. Trying to terminate it again, it told me that the process could not be terminated because it is already exiting. Even 5 minutes later, same thing. I noticed that it only had an ultra-small number of CPU cycles every few seconds, and that was it. No I/O, no change in memory. 15 minutes later. The process was still "exiting" supposedly...

Second time...well... simply opened it and that was it.
 I tried running it on two different OS environments (but on the same PC). Seems to only be a problem in the latest version.

Log
---
Rufus x86 v3.18.1877
Windows version: Windows 10 (Unknown Edition 0xBF), 64-bit (Build 19044.4170)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x043E (ms-MY)
Will use default UI locale 0x043E
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ms-MY'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
NOTE: The Malay (Bahasa Malaysia) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
0 devices found
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'Anhdv_Boot_Premium'
  Size: 3.7 GB (Projected)
  Uses: Grub4DOS
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
Using image: Anhdv_Boot_Pre_24.1.iso (3.7 GB)
CAUTION: Listing of non-USB removable drives enabled
By using this unofficial cheat mode you forfeit ANY RIGHT to complain if you lose valuable data!
0 devices found
Listing of non-USB removable drives disabled
0 devices found
0 devices found
0 devices found
CAUTION: Listing of non-USB removable drives enabled
By using this unofficial cheat mode you forfeit ANY RIGHT to complain if you lose valuable data!
0 devices found
Listing of non-USB removable drives disabled
0 devices found
CAUTION: Listing of non-USB removable drives enabled
By using this unofficial cheat mode you forfeit ANY RIGHT to complain if you lose valuable data!
0 devices found
0 devices found
0 devices found

arukateru, 2024-03-24:
Just had this same problem with different USB drive and different program. After some research it seems you simply cannot exit a process that has pending I/O in Win32 so I don't think this is fixable. Though, I'm not sure why they would have pending IO to begin with ? Maybe I'm just unlucky and have two trashy USB drives? I would close this issue but idk, maybe there is some way to fix it, I am not an advanced programmer

JonnyTech, 2024-03-24:
> Seems to only be a problem in the latest version.
> Log
> Rufus x86 v3.18.1877

FYI the latest version is v4.4

arukateru, 2024-03-24:
After buying a brand new USB drive and trying a different burning program, I tried again and got a similar issue, the program just completely froze part way through, becoming unterminatable and stuck on "exiting" until I forcibly removed the drive...this time it crashed and so I decided to look in the event log and I see that my event log is FULL of this error: Application popup: Windows - Delayed Write Failed : Exception Processing Message 0xc0000222 - Unexpected parameters. The crash reason is that the application "stopped responding to Windows". Doing some research, it seems that a lot of programs don't have any way to handle this error, and I think that's why Rufus and other programs crash. I don't know why it would be happening with a brand new, name-brand flash drive. Perhaps my installation image is corrupt. Anyways, I guess that my issue more specifically now would be: handling of delayed write fail error should be implemented into Rufus to improve stability and user experience

pbatard, 2024-03-25:
> Just had this same problem with different USB drive and different program. 

Which confirms that this is not a Rufus specific issue.

> handling of delayed write fail error should be implemented into Rufus to improve stability and user experience

Rufus handles errors that are reported by the system APIs. Even when using Overlapped, there's no such thing as issuing a `WriteFile()`, getting a success code from Windows, and distrusting that success code to read (from where?!?) a different code after some ill-defined interval.

At any rate, I'll be happy to accept a patch if you want to send one. If not, and considering that you are the only person to report this issue, which you seem to encounter with multiple applications and which therefore would tend to hint not at an application problem but at an issue with your computer that you should really look into, I'm just going to ignore what seems to be a purely environmental issue on your side.

Especially, if an application like Rufus appears to get stuck whereas nobody else appear to experience a similar issue, you clearly have an I/O problem that you want to fix.

closed 2024-03-25 by pbatard