pbatard/rufus #2449: Windows 11 installation is broken since 4.2 version of rufus

Learpcs, 2024-03-24:
Tried to install Windows 11 Consumer Edition image from https://rutracker.org/forum/viewtopic.php?t=6504194 (I verified checksums). If you patient enough you can try to download it, yet I don't guarantee it's a reproducable issue.  Issue is: pressing start button when installing GPT scheme yields to crash without error whatsoever.  I'll link video to give you some idea about the issue. 

So the steps are:
1.  Choose image "en-us_windows_11_consumer_editions_version_23h2_updated_march_2024_x64_dvd_bcbf6ac6"
2. Basically don't change any options but I used the following:
3. "Image option: Standart Windows Installation"
4. Partition scheme: GPT (Other options are non-changable) 
5. Click "Start" button
6. Get crash without error 

The issue persists until 4.2 version.  Logs didn't record anything related to the error:

```
Rufus x64 v4.1.2045 (Portable)
Windows version: Windows 10 Home x64 (Build 19044.1766)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Detected installed Windows Language Pack for 0x0409 (English (English))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Found USB 2.0 device 'Generic- SM/xD-Picture USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- MS/MS-Pro USB Device' (058F:6364)
Device eliminated because it appears to contain no media
2 devices found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {E48BF4F8-D975-5E80-E691-F274E54D3749}
Max parts: 128, Start Offset: 17408, Usable = 8053029376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Ventoy'
  Detected File System: exFAT
  ID: {83137CB5-2B84-41D3-C4AD-047FADB9768E}
  Size: 7.5 GB (8018440192 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'VTOYEFI'
  Detected File System: FAT16
  ID: {EBE061D6-5843-0113-58AA-142FC3389B5E}
  Size: 32 MB (33554432 bytes)
  Start Sector: 15663064, Attributes: 0xC000000000000001
```

Video: https://github.com/pbatard/rufus/assets/34576126/b3fc0467-2c3a-4790-bb01-6fbb35628662

JonnyTech, 2024-03-24:
Unofficial modified build of Windows not provided by Microsoft are unsupported by Rufus.

pbatard, 2024-03-25:
Not able to replicate the issue.

The difference between 4.1 and 4.2+ is that we extract some additional data from the image so that we can display the various options for the Windows User Experience dialog and my guess is that your specific installation of Windows Home is either corrupted or missing some of the standard Windows APIs we need to do so.

Since you say the app crashes __after__ you press START with 4.4, I'm a bit curious why you didn't provide the log from Rufus 4.4 where you selected the image, so that I can get some data on whether Rufus already has an issue getting all the data it needs from the image.

Alternatively, regardless of whether Rurfus crashes, you can always access the log from [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview) and it would therefore be interesting to see what kind of output you get there when the crash occurs.

But really, considering that you are the only person reporting this (and there should be a large enough number of people running Rufus on Windows 10 Home), logic dictates that it's an environmental issue with your specific system.

pbatard, 2024-04-03:
No update &rarr; Closed.

closed 2024-04-03 by pbatard