pbatard/rufus #2456: Only usb disks are supported, it would be perfect if local disks could be supported

hubgha163, 2024-04-06:
Only usb disks are supported, it would be perfect if local disks could be supported

JonnyTech, 2024-04-06:
I assume, from the lack of required information, that you did not read the FAQ as instructed when creating this issue.

Does this entry answer your question?

https://github.com/pbatard/rufus/wiki/FAQ#user-content-Power_keysCheat_modes

> Alt-F - Enable fixed disk detection (v1.3.4 or earlier)/Enable USB HDD detection (v1.4.0 or later):
This is the same as enabling List fixed (non flash) or unpartitioned USB disks (v1.3.4 or earlier) or List USB Hard Drives (v1.4.0 or later) in the advanced options. This UNSUPPORTED mode will let Rufus detect and format USB drives such as USB HDDs, which it doesn't do by default. The reason this is disabled by default is that it can be a potentially risky operation when Rufus is used by non technical-savvy people. For instance, if someone also keeps an USB HDD as a backup drive, and plug an USB Flash Drive, they may inadvertently end up formatting their HDD and lose valuable data. If you use Rufus with this option enabled, you are on your own!

pbatard, 2024-04-06:
Also https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_support_internal_HDDsSSDs

closed 2024-04-06 by pbatard