pbatard/rufus #2459: Cannot download ISO using Rufus

GrPK, 2024-04-13:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello! Whenever I try to download an ISO, this error pops up: "The remote name could not be resolved: 'vlscppe.microsoft.com". I searched if anyone else got it but it seems not

Log
---

[rufus.log](https://github.com/pbatard/rufus/files/14967189/rufus.log)

JonnyTech, 2024-04-13:
Do you have any blocklists installed? Several have that URL included, ie https://github.com/notracking/hosts-blocklists/issues/834

GrPK, 2024-04-13:
Yep, you're right, my NextDNS setting was blocking the URL. I turned it off temporarily and now it works. Thanks a lot for the suggestion!

closed 2024-04-13 by GrPK