pbatard/rufus #2461: Persistent Partition not working when using a NTFS base image.

dellamvonk, 2024-04-15:
Issue description
-----------------
We have a image that need to be used for checking GPU health. Additional we want to have the the log files on a different partition together with some health check scripts of our own.  That extra partition is created using the persistent partition option. 
Initially it worked fine since we were using version 3.20. When upgraded to the latest it did not work anymore. 
Testing revealed this has been broken since version 3.21 of Rufus. 
We suspect this fix has been responsible: 
    Fix UEFI:NTFS partition not being set in MBR mode for ISOs with a >4GB file (UwUntu)
When using a FAT32 based image it does work fine.
For now we do not have the ability to change the provide NTFS based image to FAT32. So, we like to have the persistent partition option working for NTFS again.

Thanks in advance,
Alex Vonk

Log
Rufus x86 v3.21.1949
Windows version: Windows 11 Enterprise, 64-bit (Build 22000.2836)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: Windows VDS is unavailable
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Using 'autorun.inf' label for drive F: 'Fedora-WS-Live-39-1-5'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3879, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x005A5240
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 19.7 GB (21171806208 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 10.0 GB (10737409536 bytes)
  Start Sector: 41353232, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Support-Live-Image'
  Size: 7.4 GB (Projected)
  Has a >4GB file
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: sli-0.5-rocky-8.9-xehealthcheck.iso (7.4 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\vonka\AppData\Local\Rufus\syslinux-6.04\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Warning: Could not delete partition(s): VDS is not available
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Grub 2.0 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 29.7 GB)
Write error [0x000001B1] A device which does not exist was specified.
Retrying in 5 seconds...
● Creating UEFI:NTFS Partition (offset: 31909199872, size: 1 MB)
Writing UEFI:NTFS data...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for shared write access
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{837de414-f0e4-11ee-ac40-300505b1f499}\
Disabling file indexing...
Installing Syslinux 6.04...
Opened \\?\Volume{837de414-f0e4-11ee-ac40-300505b1f499} for shared write access
Using existing 'C:\Users\vonka\AppData\Local\Rufus\syslinux-6.04\ldlinux.sys' ✓
Using existing 'C:\Users\vonka\AppData\Local\Rufus\syslinux-6.04\ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{837de414-f0e4-11ee-ac40-300505b1f499}\ as F:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: F:\EFI\BOOT\BOOTX64.EFI (921.9 KB)
Extracting: F:\EFI\BOOT\fonts\unicode.pf2 (2.4 MB)
Extracting: F:\EFI\BOOT\grub.cfg (1.9 KB)
Extracting: F:\EFI\BOOT\grubx64.efi (2.1 MB)
Extracting: F:\LiveOS\squashfs.img (7.3 GB)
Extracting: F:\isolinux\boot.cat (2 KB)
Extracting: F:\isolinux\efiboot.img (6.0 MB)
Extracting: F:\isolinux\initrd0.img (75.1 MB)
Extracting: F:\isolinux\isolinux-voltagepark.cfg (3.1 KB)
Extracting: F:\isolinux\isolinux.bin (38 KB)
Extracting: F:\isolinux\isolinux.cfg (1.9 KB)
Extracting: F:\isolinux\ldlinux.c32 (113.4 KB)
Extracting: F:\isolinux\libcom32.c32 (176.4 KB)
Extracting: F:\isolinux\libutil.c32 (22.3 KB)
Extracting: F:\isolinux\macboot.img (12.9 MB)
Extracting: F:\isolinux\vesamenu.c32 (26.2 KB)
Extracting: F:\isolinux\vmlinuz0 (10.4 MB)
Created: F:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico
NTFS Fixup (Checkdisk)...
Volume label is Support-Live-Image.

Stage 1: Examining basic file system structure ...
  256 file records processed.
File verification completed.
 Phase duration (File record verification): 8.14 milliseconds.
  0 large file records processed.
 Phase duration (Orphan file record recovery): 0.21 milliseconds.
  0 bad file records processed.
 Phase duration (Bad file record checking): 0.20 milliseconds.

Stage 2: Examining file name linkage ...
  288 index entries processed.
Index verification completed.
 Phase duration (Index verification): 12.90 milliseconds.
 Phase duration (Orphan reconnection): 0.33 milliseconds.
 Phase duration (Orphan recovery to lost and found): 0.01 milliseconds.
  0 reparse records processed.
  0 reparse records processed.
 Phase duration (Reparse point and Object ID verification): 0.42 milliseconds.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
 Phase duration (Security descriptor verification): 10.31 milliseconds.
  16 data files processed.
 Phase duration (Data attribute verification): 0.20 milliseconds.

Windows has scanned the file system and found no problems.
No further action is required.

  31160303 KB total disk space.
   7812308 KB in 27 files.
        80 KB in 18 indexes.
     44167 KB in use by the system.
     42544 KB occupied by the log file.
  23303748 KB available on disk.
      4096 bytes in each allocation unit.
   7790075 total allocation units on disk.
   5825937 allocation units available on disk.
Total duration: 33.39 milliseconds (33 ms).
NTFS Fixup completed.

Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Using 'autorun.inf' label for drive F: 'Support-Live-Image'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3879, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0071143E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 29.7 GB (31908151296 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 62322656, Boot: No

Diskmgmt image after the stick has been created. The UEFI partiton is there but partition 1 is taking up the rest of the space and partition 2 is NOT created.
[image]

dellamvonk, 2024-04-15:
[image]
Correct image

pbatard, 2024-04-15:
Please try the ALPHA you will find in the MinGW artifacts from https://github.com/pbatard/rufus/actions/runs/8628239450 and see how it goes. The issue you report should have been fixed in 0f23c4718421c8bd7dd4810b7ef759267d535f83.

dellamvonk, 2024-04-15:
Hello Pete,

Thanks for the quick response and solution 😊.
Sorry I missed the entry in the “ [INTERNAL] List of rufus-4.5 enhancements and fixes” list.

This version is working as it should. Here are the results of the quick test I ran. The text in the image is Dutch since this version was not digitally signed I had to ran it on a different computer which only talks Dutch.
-------- LOG Start
Rufus x64 v4.5.2128 (Alpha)
Windows version: Windows 11 Pro x64 (Build 22631.3155)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0413 (nl-NL)
Will use default UI locale 0x0413
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'nl-NL'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Using 'autorun.inf' label for drive D: 'Support-Live-Image'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3879, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0071143E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 29.7 GB (31908151296 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 62322656, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Support-Live-Image'
  Size: 7.4 GB (Projected)
  Has a >4GB file
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: sli-0.5-rocky-8.9-xehealthcheck.iso (7.4 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\amvc11\AppData\Local\Rufus\syslinux-6.04\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1<file://./PhysicalDrive1> for shared write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 17.7 GB)
● Creating Linux Persistence Partition (offset: 19024266240, size: 12 GB)
● Creating UEFI:NTFS Partition (offset: 31909183488, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
[... 105 lines cut ...]
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 17.7 GB (19023200256 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 12 GB (12884917248 bytes)
  Start Sector: 37156770, Boot: No
Partition 3 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1064448 bytes)
  Start Sector: 62322624, Boot: No

---------- LOG End
Schijf 1 is the USB stick.

***@***.***

Thanks,
Alex Vonk

Internal Use - Confidential

closed 2024-04-15 by pbatard