pbatard/rufus #2462: Remove Kaspersy from list of unsupported isos

solomoncyj, 2024-04-15:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
kaspersky has included a correct grub version in their iso and has also told users o use it
https://support.kaspersky.com/krd18/howto/14226#block1
[image]

Log
---
```
Rufus x64 v4.4.2103
Windows version: Windows 10 Pro x64 (Build 19045.4170)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'WD My Passport 25E1 USB Device' (1058:25E1)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
Using 'autorun.inf' label for drive D: 'Kubuntu 23.10 amd64'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1886, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {926A1BF2-7614-47DE-B758-5116C172C2A3}
Max parts: 128, Start Offset: 17408, Usable = 15518890496 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {772FEA0F-3147-4351-A94A-28FB03ABF289}
  Size: 6.5 GB (6927908864 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Linux Data Partition
  Name: 'Linux Persistence'
  Detected File System: ext3
  ID: {5AB98B12-067B-4534-A196-9C2F4D8B0CF2}
  Size: 8.0 GB (8589934080 bytes)
  Start Sector: 13533120, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Reported Grub version: 2.02
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'KRD'
  Size: 683.5 MB (Projected)
  Uses: GRUB2 (2.02)
  Uses: EFI
Using image: krd.iso (687.1 MB)

```

pbatard, 2024-04-15:
I don't understand what you are asking.

Do you see an issue with Kaspersky images? If so, what is it exactly? If not, then I don't see anything to do here.

solomoncyj, 2024-04-15:
i just letting you know you can remove it from tht "unsupported iso list"

solomoncyj, 2024-04-15:
https://github.com/pbatard/rufus/wiki/FAQ#incompatible-images

pbatard, 2024-04-15:
Removed, thanks.

Please bear in mind that people may place things in the (long) FAQ so that they don't have to keep them in mind, so pointing to the precise section you want to see altered does help.

closed 2024-04-15 by pbatard