pbatard/rufus #2464: feature request mark bad blocks unusable

gfanini, 2024-04-18:
﻿

Checklist
---------
- [ X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Why won't rufus mark as unusable detected bad blocks, which should be supported by exfat, ntfs and other file systems ?
which programs allow to exclude bad sector ?

Log
---
```
<FULL LOG>
```

pbatard, 2024-04-18:
> Why won't rufus mark as unusable detected bad blocks

Because it's not its purpose. The bad blocks check is there to tell you if your drive is defective and nothing else. That's why it is set to stop early if it detects too many bad blocks.

Also, the bad blocks detection is not performed with a specific file system but for the actual **low level** device, where there is absolutely no way to indicate which blocks are bad.

To accomplish what you want, Rufus would have to keep a list of bad blocks somewhere, and then figure out a way to map the disk blocks to the actual file system blocks, which is next to impossible on modern hardware, since it regularly remaps blocks internally, which means that by the time you have figured out that `low level disk block 1234` should map to `file system cluster 578`, the internal controller of the device may have remapped the bad block somewhere else.

In short: It is not that simple (or even that possible at all) to accomplish, and that is not the goal of Rufus anyway. Instead our advice is that, if you have a device with bad blocks, you should stop using it, and, because it values your data, Rufus will **never** help you try to use a device where bad blocks have been detected.

closed 2024-04-18 by pbatard

gfanini, 2024-04-18:
It would seem tools like these don't detect usb flash key bad blocks at low level like rufus, seen as high level clusters/sectors, 
https://www.seagate.com/it/it/support/kb/how-to-use-seatools-for-windows-202435en/
because I notice rufus specifies which type of nand test strategy ?

pbatard, 2024-04-18:
Again, I am not going to help you use a defective drive. Either you use the Rufus bad blocks test, find that it reports no error, and happily continue to use a drive that should be in fully working condition, or Rufus reports error, and in which case you are 100% on your own if you want to continue to use that drive.

Therefore, it makes no sense for Rufus to attempt do more on the bad blocks check than what it already does.

gfanini, 2024-04-18:
you seem to not understand or not want to understand my question whether rufus detects errors other programs won't ?

pbatard, 2024-04-18:
I have no idea how other programs perform the bad blocks check, and I have no interest in investigating. As such I can't tell you whether Rufus does XYZ whereas other programs do ABC. All I can tell is that Rufus tests at the block level for each individual block of the disk, which is what you want in a bad blocks __detection__ application, which is all the Rufus bad blocks check is.

If that still doesn't satisfy you, then please invest your own time to investigate differences between applications, because it's not my job to tell you about what other applications do.