pbatard/rufus #2470: rufus says it cannot patch/setup files

Bovey12, 2024-05-01:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log

pbatard, 2024-05-01:
What happens if you reboot the computer? Do you still see the `[0x00000020] The process cannot access the file because it is being used by another process.` in the log.

Basically, either you had a previous instance of Rufus that got interrupted, or you have another process, such as a security solution, that is interfering with Rufus' processing. Almost all instances of error `0x00000020` I have seen with Rufus have had to do with external software that is trying to access data it shouldn't access, thereby preventing Rufus from completing its job.

So if you still see the error after reboot, you may want to check that you're not using any of the software listed [here](https://github.com/pbatard/rufus/wiki/FAQ#incompatible-software) which is known to be incompatible with Rufus, knowing that this list is far from exhaustive...

pbatard, 2024-05-06:
No update &rarr; Closed.

closed 2024-05-06 by pbatard