pbatard/rufus #2471: estou tendo problemas com o rufus podem me ajudar porfavor

Bernardodo, 2024-05-02:
Rufus x86 v3.20.1929
Windows version: Windows 10 Home, 64-bit (Build 17134.112)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0816 (pt-PT)
Will use default UI locale 0x0816
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-PT'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7536, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002C76FE
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 57.7 GB (61990764544 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CHECKRA1N0.12.6WINDOWS'
  Size: 229.9 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Checkra1n0.12.7 - iOS 17.iso (230 MB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Adilson Cacite\AppData\Local\Rufus\syslinux-6.04\20191223\' for Syslinux installation

pbatard, 2024-05-02:
I'm afraid I'm not seeing any issue with Rufus from the log.

If you have an issue installing Checkra1n, then please post your question in the Checkra1n user forums.

closed 2024-05-02 by pbatard

emanuellatuque, 2024-05-23:
tudo bem? você comprou esse programa em algum canal do youtube? se sim eu também, estou tendo o mesmo problema acho que possa ser o pendrive que estou utilizando, ele não e original, vou testar com um original e volto para falar o resultado.

maxhammerbr1, 2024-08-10:
eu estou com o mesmo problema . meu pendrive é original . ja fiz de tudo e nada. consegui resolver?