pbatard/rufus #2474: Doc: Windows 11 Secure Boot preinstalled laptops

rb0022, 2024-05-06:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- N/A I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- N/A The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------

Documentation improvement required for Windows 11 Secure Boot preinstalled laptops. While the current documentation provides reasoning, it does not specifically document keywords and clear instructions on the options for an end-user.

Cite issue #2137 which provides some context on the current state of laptops preinstalled with Windows 11, notably the Microsoft Surface series. However this appears to be a growing trend, and while it is not possible to cover all personal computing vendors - an effort should be made to provide clear information to end-users.

---

Initial draft of suggested documentation
-----------------

Unfortunately, Microsoft has determined mulitple tiers of the "Universal" [UEFI Secure Boot Specification](https://uefi.org/specs/UEFI/2.9_A/32_Secure_Boot_and_Driver_Signing.html) which has an impact across many personal computing vendors.

The UEFI Boot Configuration of many vendors will show a similar list of options to the following:

1. Microsoft Windows only
2. Microsoft UEFI Certification Authority (including 3rd Party CA)
3. None

Examples such as Samsung Galaxy Book2/3 Pro devices, provide Secure Boot Control on/off and a separate selection for `Secure Boot Certificate Keyset`. These examples are infrequently documented and will vary for each device (even for devices using common firmware from American Megatrends International), and only screenshots of these devices will show what terms are used. However the list is explicitly shown in Microsoft Surface documentation:
https://learn.microsoft.com/en-us/surface/manage-surface-uefi-settings#uefi-security-page

For Hyper-V hypervisor Virtual Machines, this is similar with the addition of 'Linux Shielded VM Template', as seen in Hyper-V documentation:
https://learn.microsoft.com/en-us/windows-server/virtualization/hyper-v/learn-more/generation-2-virtual-machine-security-settings-for-hyper-v#secure-boot-setting-in-hyper-v-manager

For end-users with new devices seeking to use Windows 11 BitLocker, the latest BitLocker encryption requires TPM 2.0 which requires Secure Boot to be enabled. Therefore if an end-user wishes to encrypt their data, Secure Boot must be enabled and this may stop boot of any OS installation created using Rufus.

If the device is new with no data and not yet encrypted, it is recommended to first perform UEFI Configuration and select the option that does not limit the device to only Microsoft Windows (such as Option 2 above, but the description may differ for each vendor).

If the device has already been encrypted, it is recommended to check and save the BitLocker Recovery Key before taking any action. There are various options available here, and it is likely the BitLocker Recovery Key is also synchronised to a Microsoft Account if the laptop is not using a Local Account only. Expect to enter the BitLocker Recovery Key after changing the Secure Boot configuration. For more information, please see:
https://support.microsoft.com/en-us/windows/finding-your-bitlocker-recovery-key-in-windows-6b71ad27-0b89-ea08-f143-056f5ab347d6

As Rufus 3.17 or later uses [UEFI:NTFS with Secure Boot signed](https://github.com/pbatard/rufus/tree/master/res/uefi), any OS created should now boot with the device and Windows 11 should now boot with BitLocker encryption.

Please refer to Rufus FAQ ["Why do I need to disable Secure Boot to use UEFI:NTFS?"](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_do_I_need_to_disable_Secure_Boot_to_use_UEFINTFS) which describes the history of Rufus and Secure Boot.

For reference purposes, example images are shown below.

**Samsung Galaxy Book3 Pro:**

[image]

**Microsoft Surface Book:**
[image]

**Hyper-V:**
[image]

pineapple63, 2024-05-18:
It appears dell has yet another variation on the secure boot settings (athough it appears dell devices may have the CA enabled by default, or at least the device this screenshot came from had it enabled out of the box)
[image]

With dell devices, there is another (minor) thing i ran into which may hinder an attempt to reinstall windows (this is technically not a secure boot issue, but could prevent the internal drive from being detected by the installer), Dell devices seem to have RAID enabled out of the box, even for devices with just a single SSD

pbatard, 2024-08-20:
After consideration, I don't think that the Rufus FAQ is the right place to delve into the two-tier Secure Boot system and its implementation by various manufacturers.

It would be better handled by a separate FAQ page, but, to be honest, I don't really want to have to maintain such a page.

Instead, I'd much rather see it handled by sites that are **already** de-facto authoritative references about Secure Boot and Secure Boot configuration such as www.rodsbooks.com/efi-bootloaders/secureboot.html (that has a full section about differences in Secure Boot config from various manufacturers). So you may want to contact Rod Smith and see if he is willing to expand his pages on Secure Boot (which I can then send people to when they have a question in the context of Rufus).

The root of the matter is that, if I start maintaining a dedicated page on the two-tier Secure Boot system here, I will want to do it well, and this will start taking time away from other matters, which I'd rather not do (even if someone external promises that they would be the one maintaining it, and that I'd simply have to approve and push changes).

As a consequence, as much as I agree that it would be useful, I have elected **not** to do so in the Wiki pages I maintain for this project.

I hope that you can understand that decision.

closed 2024-08-20 by pbatard

rb0022, 2024-11-06:
@pbatard I understand the reasoning for the close of the PR. Please read below.

To be verbose - the draft I provided was `never` to list different manufactures UEFI, and was only intended to document the background that is dispersed in various GH Issues but with the ultimate aim of showing a new Windows 11 laptop (that requires Secure Boot) can use Rufus.  See below an alternative shorter style for the FAQ.

----

= Existing Features

== Why do I need to disable Secure Boot to use UEFI:NTFS?
....
....

== How to use Rufus and Windows 11 laptop with Device Encryption?

A new laptop using Microsoft Windows 11 enabled with Device Encryption (aka. BitLocker with TPM 2.0), must use Rufus 3.17 and later which contans the [UEFI:NTFS bootloader with Secure Boot signed](https://github.com/pbatard/rufus/tree/master/res/uefi); it is possible Microsoft may later invalidate the signature without notice.

A change is required on the UEFI Boot Configuration, therefore it is recommended to create a [BitLocker Recovery Key](https://support.microsoft.com/en-us/windows/finding-your-bitlocker-recovery-key-in-windows-6b71ad27-0b89-ea08-f143-056f5ab347d6) before any action and save the key externally, to ensure Windows can be accessed if there is an issue at any time.

Once Rufus has created the bootable drive, boot the laptop to your UEFI and change from "Microsoft Windows only" to "Microsoft UEFI Certification Authority (including 3rd Party CA)"; these descriptions change per laptop manufacturer, the example given is from [Microsoft Surface laptop UEFI](https://learn.microsoft.com/en-us/surface/manage-surface-uefi-settings#uefi-security-page).

Please refer to previous FAQ entry ["Why do I need to disable Secure Boot to use UEFI:NTFS?"](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_do_I_need_to_disable_Secure_Boot_to_use_UEFINTFS) which describes the history of Rufus and Secure Boot.

pbatard, 2024-11-07:
I'm sorry but I'm seeing nothing in the data you provide but something that users are expected to know or be able to find out easily, and that therefore requires a specific FAQ entry from Rufus.

Plus, we're at version 4.6 now, and the expectation is that people will use that version or newer, and that if they attempt to use an older version like 3.17-, they are 100% on their own, so adding an entry now, that specifically references usage of old versions, is not something I want to do.

>  it is possible Microsoft may later invalidate the signature without notice.

No.

To invalidate the signature, Microsoft would either have to revoke all third party bootloaders, or there would need to be a public CVE vulnerability reported against a specific UEFI:NTFS bootloader, that would warrant that bootloader's signature to be added in the UEFI revocation list. And while Microsoft are well known for abusing the Secure Boot signing stewardship they have been entrusted with by the UEFI committee, they are not crazy and have (and will) only do so in a manner that is not openly conspicuous and that wouldn't give rise to a legal battle that would be impossible for them to win. And revoking bootloaders, without providing any justification, is not something that Microsoft can win. Microsoft knows that. And I know that. And you'd better believe that there's nothing I'd like more than jump at the opportunity to sue Microsoft, if they ever try to overextend their Secure Boot control in a manner that can legally be used to remove the Secure Boot signing stewardship from them altogether. I am not kidding.

So, unless they want to commit to a big mistake, and see the whole Secure Boot signing stewardship taken away from them **as it should be**, they are **not** going to revoke UEFI:NTFS bootloaders without a formal security justification for doing so, and without providing an explicit warning.

Which means that, all in all, the only data that a user may need from your proposed FAQ entry is that they might have to switch from Microsoft-only to Microsoft and 3rd party in their Secure Boot configuration, which is an easy search if you run into a Security Violation and does not, in my view, require a FAQ entry from Rufus.