pbatard/rufus #2476: New option - disable encryption

ExaFlop1, 2024-05-10:
According to the new M$ policy for encrypting the device, please add to the Rufus an option to switch on/off the encryption:
https://www.youtube.com/watch?v=kTLkejC4PbI&t=630s

pineapple63, 2024-05-10:
Rufus already has an option to disable encryption

pbatard, 2024-05-10:
Yes, as far as I know, the _Disable BitLocker automatic device encryption_ that is already present is what you are after.

At any rate, [I **DON'T** look into Insider build features](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_Insider_builds_NOT_SUPPORTED), because Microsoft has a habit of completely reworking or dropping them by the time of the actual release.

closed 2024-05-10 by pbatard