pbatard/rufus #2478: win 11 install grief

m3vuv, 2024-05-20:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
the first time i made a win 11 image i set the rufus to make the usb drive in uefi mode,i set my bios to uefi boot as well,i then got an error screen saying you are trying to install in mbr mode it wont work!,both my pc and the usb drive were in ueif mode,i then set my pc to mbr mode and redid the usb as mbr same error but the oposit,ie you are trying to install in uefi mode when i was in mbr,no mater which way i do things i get the same error,please help,my pc is an hp elitebook 8730w 6 gig ram+ 2tb ssd.
Log
---
```
<FULL LOG>
```

thecatontheceiling, 2024-05-20:
https://github.com/pbatard/rufus/wiki/FAQ#error-bioslegacy-boot-of-uefi-only-media

```
If this computer supports UEFI, go to your UEFI settings
               and lower or disable the priority of CSM/Legacy mode.
```

pbatard, 2024-05-20:
It's what @thecatontheceiling says.

Windows 11 cannot be installed in BIOS/Legacy mode, and, even if your computer supports UEFI, you are **defintely** booting in BIOS/Legacy mode rather than UEFI, which prevents Windows 11 from being able to install.

There are priority settings in your UEFI firmware that decide whether a USB drive should be booted in UEFI mode or in BIOS/Legacy mode, and, right now, your settings are such that only BIOS/legacy mode is being tried for USB boot.

Unfortunately, we cannot tell you precisely what these settings are, as they are different between manufacturers and computer models (and even if you told us you computer model and manufacturer, we wouldn't know, because we don't supersede manufacturer support), so you will have to figure it out on your own.

Chances are that you are far from the only person using your computer model that had to change the priority of Legacy vs UEFI boot for USB on that specific model, so  you should either do a search (the manual from your motherboard, that usually covers BIOS settings should be a good start) or head to user forum for your type of computer or ask your manufacturer for support.

m3vuv, 2024-05-20:
my bios ueif setting choices are enabled or dissabled,the only other things
related are the sata mode etc,nothing anywhere about legacy bios etc!.

m3vuv, 2024-05-20:
the lappy is an hp elitebook 8730w with the latest bios rom flash.

pineapple63, 2024-05-20:
Its possible your system could be one of the earlier EFI systems that doesn't support booting in pure EFI mode

m3vuv, 2024-05-21:
> There is a trick i know of where you can use a Windows 10 ISO and swap the
> install file out for a Windows 11 install file (the easiest way of doing it
> would be to use Rufus to put the Windows 10 ISO on the USB, mount the
> Windows 11 ISO as a virtual disc, delete the install file from the USB, and
> copy the install file from the Windows 11 ISO to the USB)
> Ive never actually tried doing it myself, but it does apparently work
> One thing to be aware of is trick will bypass ALL of the Windows 11
> requirements, including the UEFI requirement, but a future update could
> break compatibility (i am aware of an update in the works that will break
> compatibility with some unsupported hardware, if microsoft go ahead with it)

i did try rufus but it just halts and throws errors about mbr eufi modes
etc,that was the reason for my post.,regards Paul.

pbatard, 2024-05-21:
> but it just halts and throws errors about mbr eufi modes

Because Windows 11 cannot be installed on BIOS/Legacy hardware without trickery that Rufus can **NOT** support.

So, either you figure out how to install Windows 11 in UEFI mode on your hardware, or you use a different approach, which I cannot support here, to install it in BIOS/Legacy mode. Rufus cannot be used to install Windows 11 on BIOS/Legacy hardware.

thecatontheceiling, 2024-05-21:
> so what is the solution to this?,ii just want win 11 installed,i did have
> it on my lappy a few days ago but cant remember how i did it.
> 
> 
> > Its possible your system could be one of the earlier EFI systems that
> > doesn't support booting in pure EFI mode

...I directly linked you to the **exact sentence** in the FAQ that addresses your problem. Pete also wrote a multi-paragraph explanation.

maingocvinh, 2024-05-25:
rufus block my usb boot if it created for uefi installation. but my laptop is hp and it require to turn on legacy mode to use usb boot. both uefi and legacy are turn on in my laptop bios and rufus think I can't install uefi if I use legacy mode because the legacy option is turn on. but I can install both legacy or uefi, the method I create usb boot decide which type of boot mode the OS will install. So rufus shouldn't block user to boot, if it boot fail the computer will restatt automatically so you don't have to block user. https://h30434.www3.hp.com/t5/Notebook-Software-and-How-To-Questions/USB-boot-option-is-not-available-in-BIOS-setup-can-it-be/td-p/5713940
if I leave the uefi mode only, my computer only show the hard disk with uefi os installed. I want rufus have a option for who sure and still want try to boot.

pineapple63, 2024-05-25:
> rufus block my usb boot if it created for uefi installation. but my laptop is hp and it require to turn on legacy mode to use usb boot. both uefi and legacy are turn on in my laptop bios and rufus [...]

Rufus does have an advanced option to create USBs that can boot in either mode (BIOS or UEFI)
If i remember correctly you have to press ALT + E to enable this

pbatard, 2024-05-25:
> rufus block my usb boot if it created for uefi installation

Rufus doesn't block anything. It creates the USB for the target system specified under _Target system_. If that doesn't match the target system or mode you want to boot from, then you need to make sure that the option specified under _Target system_ is the one you want.

> and rufus think I can't install uefi if I use legacy mode

You do know that BIOS/Legacy and UEFI are __incompatible__ right? So it's not "Rufus" that thinks you can't install UEFI if you use Legacy mode. It's the hardware itself that can't do it. Either you use pure UEFI for installation or you use BIOS/Legacy/CSM. You cannot use both.

Please educate yourself on BIOS/Legacy and UEFI, as you will find that, contrary to what you seem to believe, Rufus does not "Block" anything and what it does makes sense.

pbatard, 2024-05-30:
Since I don't see anything more that can be added to this issue, I will close it.

closed 2024-05-30 by pbatard