pbatard/rufus #2480: [0xC00305B4] This operation returned because the timeout period expired.

LobnieYT, 2024-05-25:
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.3593)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
1 device found
Could not access first GUID volume: [0x00000012] There are no more files.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {00000000-0000-0000-0000-000000000000}
Max parts: 128, Start Offset: 17408, Usable = 15676176896 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {F604B35B-7559-4033-8B81-880525D076A4}
  Size: 14.6 GB (15674060800 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {99FB0A53-1CDE-4DF5-A98E-C79D761A7E9C}
  Size: 1 MB (1064448 bytes)
  Start Sector: 30615480, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_RU-RU_DV9'
  Detected: Windows 11 ISO (Build 22000.194)
  Size: 5.1 ГБ (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: ru-ru_windows_11_ build 22000.194 _x64_dvd_msdn.iso (5.2 ГБ)

Computing hash for 'D:\!torrent\ru-ru_windows_11_ build 22000.194 _x64_dvd_msdn.iso'...
  MD5:    f14a9ae15aabb7f109e593d34bf57830
  SHA1:   c8a17c20bb85aedeffef8bd37dc60056e3cbe101
  SHA256: 5d4a22f070864d5d1fc7032da46799ca9196715e8bdb3d07a9e65188add16e5f
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_RU-RU_DV9'
  Detected: Windows 11 ISO (Build 22621.1105)
  Size: 5.1 ГБ (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: ru-ru_windows_11_consumer_editions_version_22h2_updated_jan_2023_x64_dvd_b4f081aa.iso (5.1 ГБ)

Computing hash for 'D:\!torrent\Microsoft Windows 11 [10.0.22621.1105], Version 22H2 (Updated January 2023) - Оригинальные образы от Microsoft MSDN [Ru]\ru-ru_windows_11_consumer_editions_version_22h2_updated_jan_2023_x64_dvd_b4f081aa.iso'...
  MD5:    fe50333d13ed7a9c514c7d0a5a8b9424
  SHA1:   c0b07f31363f36f52e8db3a9cf881473caab53aa
  SHA256: e719f3f8c5d7899626da89cffa12ec11f6710337db1adfc2be03b0c22cfa7888
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Could not access first GUID volume: [0x00000012] There are no more files.
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.6 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Could not access first GUID volume: [0x00000012] There are no more files.
No logical drive found (unpartitioned?)
1 device found
Could not access first GUID volume: [0x00000012] There are no more files.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000D70AC
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 14.6 GB (15675154432 bytes)
  Start Sector: 2048, Boot: Yes
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Could not access first GUID volume: [0x00000012] There are no more files.
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 29899 (1 block = 512 КБ)
Bad Blocks: Using offset 307476 for fake device check
Bad Blocks: Writing test pattern 0x55
Cancelling
Bad Blocks: Interrupting at block 512
Bad Blocks: Interrupting at block 512
Bad Blocks: Interrupting at block 512
Bad blocks: Check failed.
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Could not access first GUID volume: [0x00000012] There are no more files.
No logical drive found (unpartitioned?)
1 device found
Could not access first GUID volume: [0x00000012] There are no more files.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Could not access first GUID volume: [0x00000012] There are no more files.
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 29899 (1 block = 512 КБ)
Bad Blocks: Using offset 460471 for fake device check
Bad Blocks: Writing test pattern 0x55
Cancelling
Bad Blocks: Interrupting at block 2816
Bad Blocks: Interrupting at block 2816
Bad blocks: Check failed.
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Could not access first GUID volume: [0x00000012] There are no more files.
No logical drive found (unpartitioned?)
1 device found
Could not access first GUID volume: [0x00000012] There are no more files.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Could not access first GUID volume: [0x00000012] There are no more files.
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.6 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Could not access first GUID volume: [0x00000012] There are no more files.
No logical drive found (unpartitioned?)
1 device found
Could not access first GUID volume: [0x00000012] There are no more files.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0010445E
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 14.6 GB (15675154432 bytes)
  Start Sector: 2048, Boot: Yes
[image]

pbatard, 2024-05-25:
```
Bad blocks: Check failed.
```

Your drive is defective. Use a different drive, as there is nothing Rufus can do about hardware issues like defective USB drives.

As long as the bad blocks check fail, the problem is with your hardware or environment.

closed 2024-05-25 by pbatard

LobnieYT, 2024-05-25:
The problem is that the flash drive worked fine until I used Rufus now, although I did and everything was fine. It can't be that the problem is with the flash drive.

pbatard, 2024-05-25:
> It can't be that the problem is with the flash drive.

Why? USB flash drives are known to be a lot more unreliable than people think, and will fail as you used them.
If the bad blocks check reports an error, your drive **is** defective, and before you want to point at Rufus as the cause of your drive becoming defective, I suggest you read [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive).

LobnieYT, 2024-05-25:
I don’t think that the flash drive could have broken so abruptly due to Rufus; right now, with God’s help, I installed the volume, and moreover, I uploaded the files there. But every time I put an iso image on a flash drive, after 17 seconds it becomes dead, and Rufus gives an error. Magic?

pbatard, 2024-05-25:
Same argument as "my appliance worked fine until it didn't". Hardware breaks. All the time. If it satisfies you to be able to point the finger at something else than bad luck, suit yourself. But you're not going to find much support when it's simply not possible for Rufus to damage hardware.

LobnieYT, 2024-05-25:
If I'm not mistaken, you're implying that flash drives can break on their own, but Rufus doesn't break flash drives at all, right?

pineapple63, 2024-05-25:
Flash drives can and do stop working on their own
I once had one start acting up, i managed to recover it, somehow, but it then became read only after i tried testing it by copying an ISO file in its entirity (not extracting it) to the USB

thecatontheceiling, 2024-05-25:
> If I'm not mistaken, you're implying that flash drives can break on their own, but Rufus doesn't break flash drives at all, right?

Yes. Extremely obvious information.