pbatard/rufus #2481: Pendrive getting unusable

itsmeshibintmz, 2024-05-25:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

The pendrive is showing an error that " Device access is denied after I press start, and I had to manually create partition and creating bootable USB won't work. First I thought it was my pendrive issue, so I changed pendrive, and same issue. So I downloaded 4.4 version and usb is getting created with Bootable ISO.

Log
---
```
<FULL LOG>
```

pbatard, 2024-05-25:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Error_0x00000005_Access_is_denied

closed 2024-05-25 by pbatard

itsmeshibintmz, 2024-05-26:
Yep I have enable Controlled Folder Access, and after disabling it, it worked, but to say without disabling controlled folder access, Rufus v4.4 created a bootable USB.

pbatard, 2024-05-26:
There were no changes between 4.4 and 4.5 that could have affected Controlled Folder Access, and it is simply not possible for Rufus to bypass it. If you observed a difference in behaviour, it was due to something external to Rufus because, no, there is no code that I can add that can "magically" make Rufus work with a setting that is designed to prevent third application from writing to USB devices (and if it ever were to work, then **clearly** it would mean that there is a bug with Controlled Folder Access that you need to report to Microsoft rather than here, as it should not, under any circumstances, let Rufus write to USB drives unless you explicitly allow it, and the Rufus application should not be able to bypass this restriction).