pbatard/rufus #2482: Error: [0xC0030570] The file or directory is corrupted and unreadable

KireinaR, 2024-05-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: 4.5.2180 (Portable)` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Whenever I try to create bootable USB with persistent storage, I get this error. But it works fine if it has no persistent partition. I have tried it on 3 USBs and have had the same result. Each time this happens, I have to use diskpart tool to clean the drive, convert to MBR and format it to NTFS/exFAT to make it reusable for regular use again.
[image]

Log
---
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Home Single Language x64 (Build 22631.3593)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'VendorCo ProductCode USB Device' (048D:1234)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5407781E
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 117.2 GB (125828071424 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'pearOS Sonoma 14 2023.12.28'
  Size: 4.1 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which may not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: pearOS_Sonoma_64bit-14-pre-release.iso (4.1 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\LENOVO\Desktop\\Rufus\syslinux-6.04\20191223\' for Syslinux installation
Timeout while retrieving conflicting process list

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Could not open GUID volume '\\?\Volume{710f2b7b-1b06-11ef-8489-dc1ba1c63b2f}': [0x00000021] The process cannot access the file because another process has locked a portion of the file.
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 95.2 GB)
● Creating Linux Persistence Partition (offset: 102206780928, size: 22 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
360448 possible inodes out of 5767169 blocks (block size = 4096)
288358 blocks (5.0%) reserved for the super user
Creating 176 inode sets: [1 marker = 2.2 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as F: after error

Found USB 2.0 device 'VendorCo ProductCode USB Device' (048D:1234)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x029CC003
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 95.2 GB (102205685760 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 22 GB (23622326784 bytes)
  Start Sector: 199622619, Boot: No
Found USB 2.0 device 'VendorCo ProductCode USB Device' (048D:1234)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x029CC003
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 95.2 GB (102205685760 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 22 GB (23622326784 bytes)
  Start Sector: 199622619, Boot: No
Found USB 2.0 device 'VendorCo ProductCode USB Device' (048D:1234)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x029CC003
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 95.2 GB (102205685760 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 22 GB (23622326784 bytes)
  Start Sector: 199622619, Boot: No
Found USB 2.0 device 'VendorCo ProductCode USB Device' (048D:1234)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x029CC003
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 95.2 GB (102205685760 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 22 GB (23622326784 bytes)
  Start Sector: 199622619, Boot: No
Found USB 2.0 device 'VendorCo ProductCode USB Device' (048D:1234)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x029CC003
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 95.2 GB (102205685760 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 22 GB (23622326784 bytes)
  Start Sector: 199622619, Boot: No

```

TCFFan123, 2024-05-26:
Damaged USB drive likely.

KireinaR, 2024-05-26:
> Damaged USB drive likely.

All 3 of them? Likely not.

pbatard, 2024-05-26:
https://github.com/pbatard/rufus/issues?q=Failed+to+create+ext3+%27lost%2Bfound%27+dir%3A+Corrupted+entry

This seems to be a sporadic issue reported by some people and which appears to have to do with Windows not liking "foreign" file systems being created.

Unfortunately, this is not something I can replicate, which makes it exceedingly difficult to try to investigate.

I would appreciate if you could post a log with the results of a bad blocks check, as some people got this issue on defective drives (and you may have connector issues rather than bad drives, which would explain repeated failures with multiple devices).

KireinaR, 2024-05-27:
> https://github.com/pbatard/rufus/issues?q=Failed+to+create+ext3+%27lost%2Bfound%27+dir%3A+Corrupted+entry
>
> This seems to be a sporadic issue reported by some people and which appears to have to do [...]

Sure I will run a bad block check.
Update: I tried it on a different computer and got the same result.

KireinaR, 2024-05-27:
[image]
I guess there are bad blocks. Any ways to fix ?

pbatard, 2024-05-27:
> I guess there are bad blocks. Any ways to fix ?

No. Change your drive(s). And be mindful that some unscrupulous vendors out there will sell fake drives (i.e. drives that have been tampered to show a larger capacity than they actually have, and that will fail when you try to access blocks beyond the capacity), so please be also mindful of cheap drives.

As your issue was external to Rufus, I will now close it and advise anybody encountering these kind of errors not to pay "lip service" to the check list, that explicitly asks you to run a bad blocks check, but to actually do it.

closed 2024-05-27 by pbatard

thecatontheceiling, 2024-05-27:
your usb drives are probably fake

KireinaR, 2024-05-27:
Thanks so much guys. Appreciate the help, I will buy some new drives then.

N5SLI, 2024-06-04:
I am having the exact same issue. I have tried several flash drives that I know are good.

pbatard, 2024-06-04:
> I have tried several flash drives that I know are good

But did you run an actual bad blocks check from Rufus?

It's not enough to "know" that your drives are good. You need hard concrete evidence that they are, which can only be accomplished through a bad blocks check that also tests for fake drive, like the bad blocks check of Rufus.

N5SLI, 2024-06-04:
> But did you run an actual bad blocks check from Rufus?
>
> It's not enough to "know" that your drives are good. You need hard concrete evidence that they are, which can only be accomplished through a [...]

Why would rufus work fine with the flash drive when I do not have any persistence?

pbatard, 2024-06-04:
> Why would rufus work fine with the flash drive when I do not have any persistence?

Because persistence accesses pretty much all of the range of the drive in terms of sectors (due to the ext formatting of the ext partitions, with inodes being created all over the place).

When not using persistence, only the beginning of the drive is accessed. So, if you happen to have a _fake_ drive with an actual capacity that is larger than the size of your ISO (as opposed to the _fake_ reported capacity, which will be much larger than actual), then everything may look good because you're not trying to access sectors past actual capacity. But persistence will try to use all of the drive, so that's precisely when failures are expected to occur, as you will be writing sectors past actual capacity.

At any rate, just test the bloody drive for bad blocks. It's not that difficult and it's the __ONLY__ way to be sure that your drive is good. Anything else is just speculation.

N5SLI, 2024-06-04:
At any rate, just test the bloody drive for bad blocks. It's not that difficult and it's the ONLY way to be sure that your drive is good. Anything else is just speculation.

I am using a 2 tb flash drive so this may take some time.

pbatard, 2024-06-04:
If your 2 TB flash drive's price was too good to be true, then it's probably fake... These go for at least $100 (as they should), and you can be darn sure that all those cheap 2 TB flash drives you can see for sale on aliexpress, that are priced around $10 or less, are 100% fake.

N5SLI, 2024-06-04:
Actually purchased on eBay. but the second drive I have is a 64gb sandisk I purchased at Walmart and it has the same issue.

pbatard, 2024-06-04:
> Actually purchased on eBay.

Not much of a warranty that it'll be better than aliexpress. Same unscrupulous sellers operate in both spaces, and since you did not talk about how much you paid, I'm going to assume that it was way less than $100.

> a 64gb sandisk I purchased at Walmart

Not much of a guarantee either. People have gotten fake drives from Walmart. I know. Because some of them complained that Rufus didn't work properly, and when they did run a bad blocks check, they found that the drive was fake.

Now, again, until you do report with the results of your bad blocks check for both drives, this is just pure speculation on my side, since I still don't have any hard evidence to go by...

N5SLI, 2024-06-18:
Sorry for taking so long to get back to you. I am a 2-way radio technician and have been really busy lately. You were correct, there were many bad blocks found. Thank you for explaining things too me. I appreciate you taking the time to explain how persistence works. Have a great day.

Mrsaucejunior, 2024-07-13:
There is no bad blocks for mine but it is still showing same error 
Or is there any other way for checking bad blocks?