pbatard/rufus #2483: 452 out of range pointer: 0x67db010 when trying to install Blufin

igor2x, 2024-05-29:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
1. On my laptop Windows 11 computer I have downloaded Rufus 4.5 Portable from https://github.com/pbatard/rufus/releases/download/v4.5/rufus-4.5p.exe
2. I have copied rufus-4.5p.exe to empty directory (no other files are in this directory).
3. Plugged in USB key.
4. Started Rufus.
5. USB key was automatically recognized. I clicked on SELECT button and pointed to `bluefin-gts.iso`, this is Blufin image (that is based on Fedora Silverblue 39). I have downloaded ISO image from: https://projectbluefin.io/ clicked on "Try Out" button. On web page selected: Hardware = desktop; Developer = No; Primary GPU: Intel (I have Intel graphic card).
6. In Rufus I have selected NTFS as file system. Rufus asked if I want to check the internet, I clicked on Yes button, then got warning all data are going to be deleted from USB key, and I agreed.
7. ISO image was burned successfully.
8. I have unplugged USB key from my laptop and plugged in into my desktop PC.
9. On desktop in BIOS I have set to boot from USB key and started booting. And very quicky (before GRUB menu is displayed) I have got the error: **452: out of range pointer: 0xd67db010. Aborted. Press any key to exit.**

Log
---
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Enterprise x64 (Build 22621.3593)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15089, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {0D2F708C-A557-40BF-9C95-A9B4F1032708}
Max parts: 248, Start Offset: 32768, Usable = 124117778944 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'ISO9660'
  Detected File System: ISO9660
  ID: {0D2F708C-A557-40BF-9C94-A9B4F1032708}
  Size: 5.1 GB (5456619520 bytes)
  Start Sector: 64, Attributes: 0x1000000000000001
Partition 2:
  Type: EFI System Partition
  Name: 'Appended2'
  Detected File System: FAT16
  ID: {0D2F708C-A557-40BF-9C97-A9B4F1032708}
  Size: 12.5 MB (13082624 bytes)
  Start Sector: 10657524, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Reported Grub version: 2.06
Disk image analysis:
  Image has an unknown Master Boot Record
[... 488 lines cut ...]
  30301683 total allocation units on disk.
  28882633 allocation units available on disk.
Total duration: 51.66 milliseconds (51 ms).
NTFS Fixup completed.

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using 'autorun.inf' label for drive E: 'bluefin-x86_64-gts'
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15089, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x118705CB
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 115.6 GB (124115697664 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1064448 bytes)
  Start Sector: 242415558, Boot: No

```

igor2x, 2024-05-29:
Now I have downloaded Ubuntu 24.04 ISO image and followed exactly the same procedure as above with Blufin ISO image and Ubuntu successfully starts from this USB key.

I assume there is something related to Blufin ISO image from log file from first post I see "File not found: 404" error downloading core.img file.

pbatard, 2024-05-29:
Please try again after deleting your existing Rufus directory (where you will find an existing `grub-2.06-nonstandard-gdie-bluefin-gts\core.img`. You can also try <kbd>Atl</kbd>-<kbd>D</kbd> in Rufus before trying again.

pbatard, 2024-05-29:
Basically, the issue is that Fedora use a more nonstandard than nonstandard GRUB 2.06, so I have to **manually** add an exception on the Rufus download website for all these nonstandard-nonstandard GRUBs, and there are only so many Fedora derivatives I can be aware of.

closed 2024-05-29 by pbatard

pbatard, 2024-05-29:
If it works, then the download of https://rufus.ie/files/grub-2.06-nonstandard-gdie-bluefin/core.img should not produce a 404.
If you still see a 404 then please try again (and make sure to delete any local file) until it works.