pbatard/rufus #2484: Others updated isos?

seba2282, 2024-05-30:
Hello, I see in last time Rufus only see Windows 10 22H2 iso v1 and Windows 11 23H2 updated from December to download. What's goin on? Where updates ISOs? If old are possible to download? Earlier was list to choose with which update download. Best Regards

pbatard, 2024-05-30:
> What's goin on?

Because it downloads from Microsoft servers **directly**, Rufus can only provide ISOs that are made freely available by Microsoft. The other ISOs you mention are not (or no longer) freely available from Microsoft, so Rufus cannot propose them. Especially, in early 2024, Microsoft removed availability to all the older Windows 11 and Windows 10 ISO downloads, as well as Windows 7 ISO downloads, so Rufus can obviously no longer make those available.

closed 2024-05-30 by pbatard

seba2282, 2024-05-30:
Thank you for informations. :)

thecatontheceiling, 2024-05-30:
You can go to https://os.click to get older variants of Windows ISO's that Microsoft has officially released to the public.

unfortunately the only way you can verify if there ISO's are genuine or not is through checking the hash of the ISO against hash lists on the internet, here's a few of them if you need:
https://files.rg-adguard.net/search
https://awuctl.github.io/mvs/

thecatontheceiling, 2024-05-30:
The only ISO's for which Microsoft publishes checksums publicly I think are the Windows 11 ISO's, nothing for windows 10.