pbatard/rufus #2485: Rufus will be ready on windows 12?

seba2282, 2024-06-01:
How in topic. Rufus will make still let's skip Microsoft account for Windows 12 and turn off automatic encryption bitlocker? 
Best Regards

thecatontheceiling, 2024-06-01:
how would Pete possibly know? Pete doesn't have access to prototype windows 12 builds (if they even exist yet) as far as I know

seba2282, 2024-06-01:
I see that insiders especially dev insider go in few future in windows 12.
I heard only, should be better and more older devices should work good and applications too. How much is true that, I don't know

thecatontheceiling, 2024-06-01:
Windows 12 isn't a thing yet. Insider preview is for previewing Windows 11 24H2, not Windows 12 lmao

seba2282, 2024-06-01:
But you were on insider dev? There faster are testing new futures which can come or not come in stable version windows and is telling about windows 12 can be and that year. Time look few on devs

thecatontheceiling, 2024-06-01:
Completely irrelevant to whether Rufus can support Windows 12 or not. There is simply no way of knowing whether Rufus will work as-is with Windows 12 or not as it literally does not exist yet.

pineapple63, 2024-06-01:
Even if the canary channel is tied to a future release of Windows (but microsoft aren't yet saying anything), running Rufus on insider builds is not supported (theres nothing stopping you from doing so, but if it doesn't work, don't expect anything to be done about it until the official public release)

seba2282, 2024-06-01:
Nice know

pbatard, 2024-06-01:
I [don't even look at insider builds](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_Insider_builds_NOT_SUPPORTED), so I'll start looking at Windows 12 is when Microsoft starts to produce releases for it.

And for the record, as much as people want to give me credit for it, **I did not invent any of the bypasses or Windows installation options that you find in Rufus**. Microsoft actually designed those to be included in the Windows installers, and either Microsoft openly documented it or other people reported it, which in turn led to me adding these features to Rufus. Especially, I did not invest **any** time trying to find hacks in the Windows installation process. So if Microsoft decides to remove any of the bypasses or features that Rufus uses for Windows 12, and especially if they decide to remove the TPM bypass, then these features will not be available from Rufus for the Windows 12 installation, because, despite what people appear to think, Rufus does **NOT** hack or patch any Windows files.

As such, I'm going to close this issue, as I cannot speculate as to what might happen with Windows 12 (just like I don't even speculate on what might happen with Windows 11 24H2).

closed 2024-06-01 by pbatard