pbatard/rufus #2486: Dev Drive on Win 11 appears as flash drive, defaults to

asheroto, 2024-06-02:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I just started testing out the Windows 11 Dev Drive feature. I noticed that when I open Rufus, the device defaults to my Dev Drive named `Code`. I realize this is because of alphabetical order, but I'm thinking this type of drive should be seen as an HDD (or virtual disk) rather than a flash drive, thus should not appear in the list. Now the interesting thing here is that I do NOT have a drive letter assigned to the drive, it's mounted in a folder path. But yet Rufus still shows it.

[image]

Dev Drive properties:
[image]

Log
---
```
Rufus x64 v4.5.2180
Windows version: Windows 11 Pro x64 (Build 22621.3593)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found VHD device 'Microsoft Virtual Disk'
Found non-USB removable device 'Samsung SSD 870 QVO 8TB' => Eliminated
Found USB 3.0 device 'VendorC ProductCode USB Device' (346D:5678)
2 devices found
Disk type: FIXED, Disk size: 10 GB, Sector size: 512 bytes
Cylinders: 1305, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {9973AC0C-5942-4875-92E9-9F9C099FF544}
Max parts: 128, Start Offset: 17408, Usable = 10737383936 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  Detected File System: (Unrecognized)
  ID: {8CA6F731-27B6-46C2-9CA7-B8C4D89CFC4E}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Detected File System: NTFS
  ID: {9B3C0A42-F039-4413-AED2-57D7740EE2E8}
  Size: 10.0 GB (10719592448 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000
```

pbatard, 2024-06-02:
1. We don't list drives in alphabetical order. We list them in order of size (so that it makes it a bit less likely for someone to erase a large backup drive when they wanted to created a small bootable USB Flash Drive)
2. Something tells me you altered the creation process of your Dev Drive because, last time we checked, NTFS was not an option, and, per ebe01cc7b691cf1014d21dbd719ed209ac5d647e, what we saw when we tested the feature (to filter it out) was that _"Microsoft Dev Drives are VHDs consisting of a small MSR followed by a large (50 GB or more) ReFS partition"_. Did you alter the default Dev Drive creation process in any way?

pbatard, 2024-06-02:
I'll also add that the fact that your main Dev drive partition is highly suspicious considering the Dev drive creation dialog that explicitly stipulates that the size should be at least 50 GB:
[image]

asheroto, 2024-06-02:
1. Gotcha, makes sense.
2. Nope, I just created it using the standard method. Here is recreating using the same method. Looks like NTFS is the default selection.

[image]

[image]

[image]

asheroto, 2024-06-02:
I do not have that verbiage. Using Windows 11 22H2 build 22621.3593.

asheroto, 2024-06-02:
Even with 55 GB it also chooses NTFS, no ReFS or FAT/FAT32 option this time.

[image]

pbatard, 2024-06-02:
Hmm, your Dev drive creation dialog is surprising as it is different than mine. And I don't have the choice for the file system.
I'm using Windows 11 23H2 22631.3672.

It would seem that the **Edition** of Windows might be a factor, which, if it's the case, will make it a real pain to try to filter Dev drives out, as it's going to be next to impossible to tell if a VHD(X) with an MSR is a Dev drive or something else that the user created themselves (such as a virtual machine VHD).

pbatard, 2024-06-02:
I'm going to have to test if I can replicate your dialog on Windows Pro vs Windows Pro for Workstation (the latter being what I use), but again, if it's an Edition issue, filtering Dev drives out might be very problematic...

asheroto, 2024-06-02:
Gotcha.

All 3 drives I created show up.

[image]

I realize I should update. Oddly enough, although I've been installing automatic updates, the system did not update to 23H2 automatically so I suppose I will need to run the W11 installer assistant.

I know that with `Get-PSDrive` it doesn't show the dev drives, so perhaps there is a way. Let me know if you want me to run any commands or beta tests.

asheroto, 2024-06-02:
Must have been a transient issue when out of date. Now seems to be fine, dev drive doesn't show. Closing this.

closed 2024-06-02 by asheroto

pbatard, 2024-06-03:
Yeah, I just tested Dev Drive creation on Windows 11 23H2 Pro, and never got a dialog like yours. Dev drive min size was 50 GB and no file system choice was proposed.