pbatard/rufus #2487: Windows to go is missing in every release

Randywfl, 2024-06-06:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [xx] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have downloaded and tried almost every release from 3.2 up to 4.5 and cannot find a single copy to offer the Windows To Go option.  I thought that was the purpose of Rufus but the option is gone. I've seen the option in various web posts so I know it did exist.  I was trying to create a Windows 7 to go on an SD card plugged into USB port using two different SD card readers using an ISO I created from my official MS DVD Windows 7 Professional that was proven to work on the internal HDD, before the onboard HDD controller went belly up.  I can still boot o n the built-in SD card reader and that is why I am attempting to resurrect this old Asus laptop using SD reader.

Log
---
```
<FULL LOG>
```
[rufus.log](https://github.com/user-attachments/files/15604255/rufus.log)

thecatontheceiling, 2024-06-06:
from your log:
```
Disk type: Removable
```
Microsoft didn't design older Windows versions to boot from drives marked as Removable, it MUST be fixed in order to work. There's no way to convert a Removable drive to a Fixed one so unfortunately it will not be possible to put Windows 7 in Windows to go mode on that SD card.

You'll likely need to use a Windows 10 or 11 ISO instead.

pbatard, 2024-06-06:
Don't double post.

closed 2024-06-06 by pbatard

Randywfl, 2024-06-06:
Sorry,  I wanted to correct one Tick where I had [xx] because I was afraid my 
post would be ignored if I 
didn't have the simple [x] and I could  not find any option for [Edit] so I clicked my back buton instead.  Chalk oit up to GitHub Noobie IO error.  So sorry.  Double strike on this keyboard is a constant irritation with no discernible solution ao I ended up with [xx]..
Thanks,

Randy Walker

Randywfl, 2024-06-06:
Yes, closed and thanks very much for the fully effective and rapid response!!!
 
Thanks,

Randy Walker

#2487)

Closed #2487 as completed.

Randywfl, 2024-06-07:
Hi Lyssa,
 
Win10 is has too much overhead for my little Old Atom Based laptop   I already 
have a win7 pro DVD & license key so I was trying to just go with that.  Forced 
into early retirement 5 years ago from brain surgery and money is too tight now 
on my meager budget to waste on unnecessary toys,  Otherwise There are other 
options that promise results where Rufus has failed me.

BTW -- Thanks for your replies :)
 
 
Thanks,

Randy Walker

#2487)

pbatard, 2024-06-07:
> I already have a win7 pro DVD & license key so I was trying to just go with that

Then go through a standard Windows 7 installation. I don't see why you think you need Windows To Go. Considering that your money is tight, I can guarantee that, even if you were able to run Windows 7 To Go off an USB device, that USB device would be way too slow for Windows to run properly, and you would realise that it was a bad idea anyway.

Besides, Windows To Go was **never** supported for Windows 7. Windows To Go as a feature was introduced with Windows 8. So, even if I had nothing better to do with all the time in the work (which I don't), and thought that spending time on anything related to Windows 7 was actually going to help anyone, including yourself (which I don't. I truly believe that people who are hell bent on still wanting to use Windows 7 in 2024, **no matter how rationale they think their reasons are** are doing more harm to themselves than they ever want to realise), I wouldn't be able to magically add Windows To Go support. I'm not Microsoft. I don't have special magic ability that allow me to materialize Windows features that Microsoft never introduced out of thin air. Some people might have (foolishly I might add) devoted time an energy to find workarounds to make Windows 7 run as Windows To Go, but, as far as I am concerned, if Microsoft did not provide the feature in the first place, it's not something I will EVER want to even remotely try to support.

And let me give you the kicker: The root of the matter is that you are being limited in what you want to accomplish because you are hell bent on trying to use a **proprietary closed source OS** while assuming that people who develop for that OS can offer you the same capability as they would with an Open Source OS like Linux (where, because one has full access to the source, a motivated developer can relatively easily provide extended support or backport features that were not available in the first place, which is not something you can even do for closed source software). So, IMO, if you want to get the most benefits of your old Atom based laptop, you should switch to GNU/Linux rather than stubbornly try to keep a sinking ship (Windows 7) afloat, as well as put yourself, and everyone you will exchange data with, at risk due to the fact that Windows 7 can not even remotely be considered secure enough, regardless of how you plan to use it. So your entire predicament is caused by wanting to use a closed source proprietary OS long past the date anyone (including the OS manufacturer) is willing to provide any support for it. Therefore, you can either take the advice of trying to ditch that OS at long last. Or you can ignore it, and waste more time, sweat and tears in trying to keep a sinking ship afloat. 

Oh, and just one last thing: Try as hard as you can, you will not manage to "guilt" other people in trying to help you keeping that sinking ship afloat.

To summarize: You **are** on a Titanic, but you are actually lucky enough that this is a Titanic with enough lifeboats. And yes, getting on a lifeboat (GNU/Linux) is a daunting process. However, if you do that, you will find that you are glad you did, rather than stay with the ship until its miserable end at the bottom of the ocean...

pineapple63, 2024-06-07:
With Windows to Go, unfortunately an SD card would be too slow for Windows to run properly, even if the Removable vs Fixed issue could be worked around
Even if you wanted to use an external USB HDD , unless the system has a USB 3 port that you can use, you would likely run into problems related to the speed of the drive being insufficient (a USB HDD uses the USB controller in the computer instead of the controller used for the internal drive, the SD card slot is likely also a USB device internally)

Randywfl, 2024-06-07:
Then go through a standard Windows 7 installation. I 
don't see why you think you need Windows To Go.
 
The laptop HDD controller died but i can still boot on the built-in SD card reader, so I have been looking for any means to get my Win7 Pro installed onto a spare 256GB Sd card I happen to have on hand already  Unlike any USB Thumb, when I insert my SD card it sits flush with the Laptop chassis so there is no protruding device as there is with a USB Thumb.  As for being slow, Its an Atom based laptop so it id already very slow and I have been accustomed to that since I got the laptop.
 
I was able to install, boot and run  Windows 7 Home Premium using an another WIn2go product using my Win7 pro DVD Iso created using the Imgburn product and it ran surprisingly fast considering it was running on the built-in SD card reader, And I would have been satisfied with that but it would not accept my official Windows 7 Professional License key that came with the DVD -- rendering the installation worthless to begin with.
 
All else failed, I am considering Ubuntu 16 as an alternative OS but Linux has and always will fall very short of Windows capabilities.  30 years experience with Linux, SCO, RedHat, Centos, so I am very familiar with Linux installlations.  As you know, MUCH easier than Windows. LOL
 
I can see I've 
really ruffled you  feathers and 
I'm sorry bout that.  Maybe there is some way I can delete my GitHub post and we can just put this whole ordeal behind us.  I will review and try.
 
 
Thanks,

Randy Walker

Randywfl, 2024-06-07:
Hi Pineapple,
 
You don't know because you've never done it so you don't know what you are talking about.
 
FYI -- I was able to install, boot and run  Windows 7 Home Premium from built-in SD card reader using an another WIn2go product using my Win7 pro DVD Iso created using the Imgburn product and it ran surprisingly fast considering it was running on the built-in SD card reader, And I would have been satisfied with that but it would not accept my official Windows 7 Professional License key that came with the DVD -- rendering the installation worthless to begin with.
 
Thanks,

Randy Walker

pineapple63, 2024-06-07:
>And I would have been satisfied with that but it would not accept my official Windows 7 Professional License key that came with the DVD -- rendering the installation worthless to begin with.>

I don't know how likely this is to be the case, but its possible the licence key may have already been used the maximum allowed number of times (athough you can transfer a product key to a different computer in the case of a retail licence (OEM licences don't let you do that at all), microsoft only let you do that so many times before they block the key)

With the HDD controller issue, i am curious as to if it is the SATA/IDE controller (which is what the motherboard uses to interface with HDDs and CD/DVD drives, SATA is a newer standard which came out shortly after Windows XP was released) thats gone bad, or if it is actually a fault with the controller board of the HDD itself (as if its the HDD itself, then it would be possble to replace it)

Randywfl, 2024-06-07:
There was no internet connection when I tried to enter the license so there is 
no way I hit any limit.  The laptop is an Asus EEE PC H1000 with Built-in 
internal SATA HDD interface.  The plan was to install Win7 pro (again) on the 
same laptop using the same Win7 Pro DVD I used to upgrade from the XP pro that 
came with the laptop using a spare SD card I already had on hand without any 
further cash investment, because I am retired, on a very limited budget.
 
 
Thanks,

Randy Walker

#2487)

Randywfl, 2024-06-08:
There's no way to convert a Removable drive to a Fixed one
 
Actually there is a way -- Google "SD to IDE adapter" or "SD to SATA adapter"
 
Only issue I have with that is not going to spend any money on this 16 yr old laptop and as mentioned earlier, controller is dead.
 
Thanks,

Randy Walker

pineapple63, 2024-06-08:
I seem to recall that other people have had problems with product keys when doing Windows to Go installations
I can only really guess, but it is possible that either Windows to Go with Windows 7 has broken activation (as it was never supposed to exist from my understanding), or that Windows to Go in general only works with Volume Licence keys, or possibly both

Regardless of what the issue is, i think ive spent more than enough of my time on this

thecatontheceiling, 2024-06-08:
> I seem to recall that other people have had problems with product keys when doing Windows to Go installations
> I can only really guess, but it is possible that either Windows to Go with Windows 7 has [...]

The activation servers have been rejecting all Windows 7 and 8 keys since 2022. Every product key that isn't for Windows 10 or 11 will fail to activate any version of Windows, as they're blocked by Microsoft on the server-side.

Randywfl, 2024-06-08:
Just an FYI -- you get many hits praising Rufus as best solution when you 
google "install Windows 7 on USB using rufus".

One such hit gave me this page praising Rufus on its ability to do Windows 7 to 
go:
https://www.ubackup.com/windows-7/rufus-create-bootable-usb-windows-7-8523.html
Then on to explain how Aomei Partition assistant could also do the same thing 
even easier, so Of course I tried both.  Aomie wants me to purchase an upgrade 
to do win7 Professional, so that option was out.  Hasleo WInToGo, isumsoft-
sysonusb and EaseUS OS2GO are the same story.. Although Hasleo WInToGo did 
allow me to burn an SD bootable into Windows Home Premium from my win7 Pro DVD, 
which I found very strange. 
 
Thanks,

Randy Walker

pbatard, 2024-06-08:
> One such hit gave me this page praising Rufus on its ability to do Windows 7 to go:
> https://www.ubackup.com/windows-7/rufus-create-bootable-usb-windows-7-8523.html

No it didn't. There is not a single instance of "To Go" on the page you linked.

Windows To Go has never been supported for Windows 7 with Rufus, only regular installs, which is what the page you linked to documents. And, since Rufus has existed for some time, and used to support Windows 7 before I dropped it completely, I can't be held responsible for the accuracy of non up-to-date old information provided by third parties I have had no contact with about Rufus.

At this stage, I'm going to lock this thread, because you are still trying to flimsily justify how your endeavour is not a complete fool's errand (if your HDD controller is failing, your ship is 100% sinking for crying out loud!) and how Rufus ought to support it, which it never will. You got as much help as you will ever get for this here. If you need more, please go look elsewhere, thank you.