pbatard/rufus #2488: Windows to go is missing in every release

Randywfl, 2024-06-06:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have downloaded and tried almost every release from 3.2 up to 4.5 and cannot find a single copy to offer the Windows To Go option.  I thought that was the purpose of Rufus but the option is gone. I've seen the option in various web posts so I know it did exist.  I was trying to create a Windows 7 to go on an SD card plugged into USB port using two different SD card readers using an ISO I created from my official MS DVD Windows 7 Professional that was proven to work on the internal HDD, before the onboard HDD controller went belly up.  I can still boot o n the built-in SD card reader and that is why I am attempting to resurrect this old Asus laptop using SD reader.

Log
---
```
<FULL LOG>
```
[rufus.log](https://github.com/user-attachments/files/15604255/rufus.log)

Randywfl, 2024-06-06:
I suppose I should add that I created my ISO using Imgburn, but I don't see how that should affect Rufus showing the Windows to go option.  And if I let Rufus burn the SD card without the windows to go option, It successfully creates a bootable Windows 7 professional installation USB which does me no goof at all, even though it runs perfectly.

thecatontheceiling, 2024-06-06:
why did you make a duplicate of the issue you made earlier?

pbatard, 2024-06-06:
> `Detected: Windows 7 ISO (Build 7600.16385)`

https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

closed 2024-06-06 by pbatard

Randywfl, 2024-06-07:
Hi Pete,
 
How very rude of me (tail between my legs).  I can see you stay very busy and I 
did not even think to thank you for taking time to respond to my post on 
GitHub.  
You've done a lot and been very generous with a lot of really great work  Thank you very much for all you have done.  Unfortunately for me I didn't 
get in early enough to take advantage of it.
 
 
Thanks,

Randy Walker

#2488)

Closed #2488 as completed.

Randywfl, 2024-06-07:
Hi Lyssa,
 
Sorry about that. GitHab noobie.  Needed to edit [xx] to be just [c] so my post 
would not be ignored, and did not see any option to edit my post so I clicked 
my back button made the edit and clicked send again.  So sorry.
 
 
Thanks,

Randy Walker

#2488)