pbatard/rufus #2490: Проблема создания

Fufelschmerc, 2024-06-06:
﻿Здравствуйте!
Установил последнюю версию Руфуса и после приложения от Microsoft начал делать переносной Флэш-накопитель на Руфусе с Виндоус, но очень долго идет разметка строения (GPT)
[image]

pbatard, 2024-06-06:
I need a log!

Please don't double post.

closed 2024-06-06 by pbatard

Fufelschmerc, 2024-06-06:
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 10 Home x64 (Build 19045.4412)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's
SKUSiPolicy.p7b
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1912, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5A5B3276
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.6 GB (15727591424 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_RU-RU_DV9'
  Detected: Windows 11 ISO (Build 22000.434)
  Size: 5.2 ГБ (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: ru-ru_windows_11_updated_jan_2022_x64_dvd_b92cd663.iso (5.2 ГБ)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.6 GB)

Fufelschmerc, 2024-06-06:
> Please run a bad blocks check, as requested in the check list. Your drive
> is probably bad.

А как это сделать?

Fufelschmerc, 2024-06-06:
>стоит ли прекратить загрузку? Если нет, то можете прислать пожалуйста команду для решения проблемы? Я просто слабо понимаю в этой теме

pbatard, 2024-06-06:
> А как это сделать?

This was described in the checklist that you chose to ignore:

> - [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.

If the process seems stuck, just kill the application or remove the USB drive.