pbatard/rufus #2493: Rufus accepts creation of local user names that Windows does not accept

nydydn, 2024-06-07:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted/attached the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is am attaching is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.
- [X] I did replace the username which is private data

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
If the user uses the option to create a windows local account named "local", everything seems to work fine and the install goes straight to asking the user to login, but the user named "local" with a blank password doesn't work. This is almost surely because Windows doesn't allow the creation of local usernames named "local". I couldn't find documentation for this anywhere, although it might very well exist somewhere. This being the case, it would be useful for Rufus to not allow its users to set the future local account for Windows to be named "local", and possibly other forbidden strings.
I verified that the issue is using "local" as windows user name by creating another usb stick that doesn't use the creation of a local user during installation, and then trying to set the user to be called "local" during installation, which prompted me with a "Something went wrong" error. The error went away when I used a different user name. Windows just doesn't want a user name to be called "local".

Log [rufus2.log](https://github.com/user-attachments/files/15742710/rufus2.log)
---
```
The log is attached as GitHub only accepts comments up to 65536 characters and only the log itself is longer than this.
```

As a side note, I found the Welcoming note for creating an issue unnecessarily rude and aggressive. I actually agree with all the fundamental points of the message. It is indeed a very good idea to instruct users to take certain steps that increase their chances to solve their issue, which isn't a software issue in itself, while at the same time decreasing the time necessary for developers to go through issues. This being said, there is absolutely no need to be rude and condescending about it. More specifically, I am referring to the "WHICH ARE ONLY THERE TO HELP *YOU*", which is not true, as the steps are there to help the developers as well. Also there is no need to threaten strangers over the internet with messages like "YOU HAVE BEEN WARNED."

pbatard, 2024-06-07:
Thanks for the report.

>  I found the Welcoming note for creating an issue unnecessarily rude and aggressive.

One day, you will perhaps be on the other end of trying to help people and allocating the maximum amount of time to do so. Which means reducing the unnecessary back and forth to the smallest amount possible. Which means making it clear that if one wants to receive help from a busy developer, they need to get ready to put in some of their time too, as this is not the one-way street of _"Just __guess__ my issue from a super minimal amount of information and from completely wild assumption I'm going to make about how, because it happened to me, it will be super easy to replicate"_ that some people seem to think it is.

The "rude and aggressive" note is actually a direct consequence of users being implicitly rude and aggressive with me, by withholding the information that I actually need to help them.

> as the steps are there to help the developers as well.

To help the developer help the user. The end of the chain is __always__ the user. Trust me, I get very little from issues, apart from low key abuse (_"How dare you ask me to do stuff and tell me that if I don't do it, you will ignore my report! A developer should just be grateful that a user bothered to log an issue regardless of the amount of useful data they might extract from it!"_). You wanna pretend otherwise, so that you can satisfy your idea that, in a 100% free application, asking users to put a minimum of effort when reporting an issue is unwarranted, fine. But that is a far cry from the reality of how much time developers end up wasting __not__ helping users, because they have to ask repeatedly for data and enter completely extraneous communication to try to squelch the preconceptions of some people, who, because they have only one side of the equation, are completely misreading the supposedly rosy landscape from other side.

> Also there is no need to threaten strangers over the internet with messages like "YOU HAVE BEEN WARNED."

The close to daily issue I have to close because people chose to ignore the check list tells otherwise. Oh, and this is public data, if you don't believe me. Just spend some time looking at the data from this issue tracker, and you will see how many people actually seem to completely disregard the instructions, and especially the requirement for a log.

So, yes, a warning is warranted and a very explicit notice __is__ warranted.

thecatontheceiling, 2024-06-11:
I work on a project which has 82k stars at the time of writing and I am the main "support" person in the official discord server. I agree with Pete here because I understand that **people do not know how to read**. The amount of people that join the server and make a support thread while ignoring **literally every single warning/notice** in _both_ the server AND the software itself is mind-boggling. We had to add the line: "If you get an error, please read the instructions written in bright blue first before asking us for help".

commit 6d29120b (2024-07-08): [wue] add 'Local' to the list of disallowed local account names

* Closes #2493.
* Also add 'KRBTGT' as it is mentioned at https://learn.microsoft.com/en-us/windows-server/identity/ad-ds/manage/understand-default-user-accounts#default-local-accounts-in-active-directory
  along with 'Administrator' and 'Guest' and is probably disallowed too.
closed 2024-07-08 by pbatard