pbatard/rufus #2496: Multi boot finally?

seba2282, 2024-06-09:
How in topic. Much people see multi boot is more practical and future-proof. Best Regards

pbatard, 2024-06-09:
> Much people see multi boot is more practical and future-proof.

And I posit that the vast majority of people do not need multiboot, as they just need to create a one-time boot media for OS installation, that they are not going to reuse and that they also will not want to dedicate as a boot drive.

In 12 years of looking at it, I still see multiboot as a niche market, that does have its very __vocal__ proponents, but that remains useful only for poweruser/sysadmin type people, which the majority of people installing an OS aren't.

This is the old fallacy of _"If I see something one way, then EVERYBODY must see it the same way"_ which doesn't translate to reality.

And before people start to push the idea that Ventoy demonstrates that there is demand, please bear in mind that the actual killer feature of Ventoy is not its ability to multiboot but its ability to just copy the ISO you need as is without recreating the media, which is indeed a very neat trick. On the other hand, there were multiboot utilities prior to Ventoy, that did not have that feature, and that, if multiboot was really that much of a killer feature, never seemed to take of, despite answering that demand.

So I stand by https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_add_multiboot_support. If you want multiboot, just use Ventoy or (if you really want 100% guarantee that your multiple ISOs will always boot) an IODD drive.

closed 2024-06-09 by pbatard

seba2282, 2024-06-09:
As I understood good, you prefer only modification general per one iso and put as modified ok usb boot. You told too something about their secure boot. I too started wonder if Ventoy seriously safe.