pbatard/rufus #2497: Issue with Rufus burning iso from Aoemi Backupper

UNC1RLM, 2024-06-09:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
i get this message when I use Rufus to burn a ampe. iso  (Aoemi Backupper). I contacted them and they said contact you about this.
[rufus.log](https://github.com/user-attachments/files/15752491/rufus.log)
[image]

Thanks for any help

BobM.
Log
---
```
<FULL LOG>
```

pbatard, 2024-06-09:
> I contacted them and they said contact you about this.

LOL.

You will need to contact them back and tell them that their ISOs (or at least the one you use, maybe they have updated them since) use a Windows bootloader that is affected by the _BlackLotus_ Secure Boot UEFI vulnerability, which is what Rufus is warning you about. Or maybe, in case it's not Aoemi Backupper that provides the ISO bootloaders, but they were picked up from the system you are trying to back-up, it's your system that is vulnerable to _BlackLotus_.

At any rate, one of these should be updated so that it doesn't use a vulnerable UEFI bootloader, and that is **not** something Rufus can fix for you.

Or, since this is a simple warning (this is not an error) you can just press OK to continue, knowing that, if you are trying to boot that media on a system that has been updated to prevent _BlackLotus_ vulnerable bootloaders from running where Secure Boot is enabled, you may get a boot error since these vulnerable bootloaders have been (rightfully) revoked as potentially unsafe.

Is the system you are trying to back up fully up to date with Windows Update? Because, if that is the case, then its bootloaders should have been updated by Microsoft to versions that are not vulnerable, and not produce this message. In which case, that would mean that the bootloaders that are producing the warning come from Aoemi Backupper, and you will need to ask them for a version of their software that use updated non-vulnerable bootloaders. Oh, and if that is indeed the case, you may also want to ask them to confirm that they have a deal to redistribute Microsoft copyrighted content, as these bootloaders are copyrighted by Microsoft, and one needs a license agreement to redistribute any Microsoft copyrighted content...

At any rate, there's really nothing more I can do for you here. Either you use an ISO that uses non-vulnerable bootloaders, or you ignore the warning. And if you want to know more about the reason why Rufus produces this warning, you can take a look at #2244, knowing that it meanders a lot around the whole subject matter.

closed 2024-06-09 by pbatard

UNC1RLM, 2024-06-09:
Yes everything is updated. This is all greek to me so I will write Aoemi..doubtful they care lol.

Thank you. greets and Best Wuishes,
Bob M.