pbatard/rufus #2498: Unable to patch/setup files for boot - 4.5.2180

n-tropy247, 2024-06-11:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When trying to use any of the popup options for a Windows ISO, Rufus completes the imaging process and then throws "unable to patch/setup files for boot". The unmodified ISO is successfully written to disk, but none of the selected modifications work.

Log
---
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Pro x64 (Build 22631.3593)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7540, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {A72D3CB8-3712-4802-B8CD-058E9668EB68}
Max parts: 128, Start Offset: 17408, Usable = 62026381824 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {1F6D342D-F503-46C6-B2A8-9B8538835E0E}
  Size: 57.8 GB (62024253440 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {67E64BCA-2E3C-44E4-99A9-4DE6EBADF97F}
  Size: 1 MB (1064448 bytes)
  Start Sector: 121143204, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: '11_23H2_RET_x64_en-us'
  Detected: Windows 11 ISO (Build 22621.2861)
  Size: 5.3 GB (Projected)
  Note: File on disk is larger than reported ISO size by 532 KB...
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: 11 23H2 Professional x64 en-US.iso (5.3 GB)
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7540, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {A72D3CB8-3712-4802-B8CD-058E9668EB68}
Max parts: 128, Start Offset: 17408, Usable = 62026381824 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {1F6D342D-F503-46C6-B2A8-9B8538835E0E}
  Size: 57.8 GB (62024253440 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {67E64BCA-2E3C-44E4-99A9-4DE6EBADF97F}
  Size: 1 MB (1064448 bytes)
  Start Sector: 121143204, Attributes: 0x9000000000000000
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
• Disable data collection
• Use 'groupeleven' for local account name
• Use the same regional options as this user's
• Disable bitlocker

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 57.8 GB)
● Creating UEFI:NTFS Partition (offset: 62025320448, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{983a9405-5c0d-4639-9adc-2108458be60e}\
Disabling file indexing...
Successfully remounted \\?\Volume{983a9405-5c0d-4639-9adc-2108458be60e}\ as F:
Extracting files...
Image is a UDF image
[EXTRACTION OMITTED FOR BREVITY - GITHUB MAX CHARACTER COUNT EXCEEDED]
Applying Windows customization:
Created 'F:\sources\appraiserres.dll' placeholder
Mounting 'F:\sources\boot.wim[2]'...
Could not mount 'F:\sources\boot.wim [2]' on 'C:\Users\GROUPE~1\AppData\Local\Temp\RufEA90.tmp': [0x000004DB] The specified service does not exist.
Finalizing, please wait...
F:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...

Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7540, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {646FAFDC-80AB-4062-9233-CBA1A19A1A78}
Max parts: 128, Start Offset: 17408, Usable = 62026381824 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {983A9405-5C0D-4639-9ADC-2108458BE60E}
  Size: 57.8 GB (62024253440 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {C1A27E31-D2AC-4BDD-AF57-6AD49148FE3D}
  Size: 1 MB (1064448 bytes)
  Start Sector: 121143204, Attributes: 0x9000000000000000
Checking for Rufus updates...
Checking release channel...
No new release version found.
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found

Computing hash for 'C:\Users\rileyc\Downloads\MediaCreationTool.bat-main\MediaCreationTool.bat-main\11 23H2 Professional x64 en-US.iso'...
  MD5:    aac77a677520d536529f300777cd3abd
  SHA1:   8829da9aabe0a263bb883f7a0091923c620f236a
  SHA256: 26f7b2efc8f8ac9f165c12b64abbb0684a405a2927f52d46bb266ea5cae06cf7

```

pineapple63, 2024-06-11:
That doesn’t look like an official ISO, so don’t expect to receive help

n-tropy247, 2024-06-11:
It was an ISO received from the MediaCreationToolBAT, but it also occurs with the unmodified ISO as downloaded from the [Media Creation Tool](https://www.microsoft.com/software-download/windows11) provided by MS:
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Pro x64 (Build 22631.3593)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'USB DISK USB Device' (24A9:205A)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x011AD286
Drive has an unknown Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 117.2 GB (125829087232 bytes)
  Start Sector: 64, Boot: Yes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 11 ISO (Build 22621.2861)
  Size: 5.3 GB (Projected)
  Note: File on disk is larger than reported ISO size by 566 KB...
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (5.3 GB)

Computing hash for 'C:\Users\groupeleven\Downloads\Windows.iso'...
  MD5:    ac31061d689047157691fb583f7d273e
  SHA1:   33354f98f90662168fdf33c509eb22df0b683a03
  SHA256: 555f64be9df73f9c0b8b62a604f4f267aa9311c729966d79aa33916da4491e21
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
• Disable data collection
• Use 'groupeleven' for local account name
• Use the same regional options as this user's
• Disable bitlocker

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 117.2 GB)
● Creating UEFI:NTFS Partition (offset: 125828011008, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
[... 963 lines cut ...]
NTFS Fixup (Checkdisk)...

Found USB 3.0 device 'USB DISK USB Device' (24A9:205A)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {87B0FB10-2820-4163-905B-E2C6758E5927}
Max parts: 128, Start Offset: 17408, Usable = 125829085696 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {82556E4C-B82C-4EC0-8BEB-CCE29A6023A3}
  Size: 117.2 GB (125826945024 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {E3A38083-AF47-49CF-BE42-CD50128E4967}
  Size: 1 MB (1064448 bytes)
  Start Sector: 245757834, Attributes: 0x9000000000000000

```

JonnyTech, 2024-06-11:
> Could not mount 'D:\sources\boot.wim [2]' on 'C:\Users\GROUPE~1\AppData\Local\Temp\Ruf8F9D.tmp': [0x000004DB] The specified service does not exist.

Are you using a non-standard TEMP folder?

thecatontheceiling, 2024-06-11:
MediaCreationTool doesn't make unmodified ISO's. The ISO has slightly different contents everytime you use MCT since it makes the ISO on your PC locally instead of downloading a pre-made one from Microsoft.

pbatard, 2024-06-11:
Most likely `[0x000004DB] The specified service does not exist.` means that your platform is missing the WIM/ESD imagine handling tools and APIs that are provided with __standard__ installations of Windows.

Did you customize your installation or remove Windows components?

Rufus requires the availability of the WIM mounting APIs to work and if you are seeing an `0x000004DB` error, it basically means that the `WIM###()` function calls that `wimgapi.dll` is supposed to provide are not available on your system...

pbatard, 2024-06-11:
Note that this may also mean that you disabled Windows Services that the system requires to use the WIM APIs, in which case, there again, you will have to figure out what these services might be since, by the looks of it, your issue appears to be purely environmental...

n-tropy247, 2024-06-11:
Understood, I'll take a look and update. My own Windows installation is unmodified but that doesn't mean it's fully functional. Thanks for the direction!

thecatontheceiling, 2024-06-11:
You can follow the steps here to do a "repair" re-installation of Windows, it should fix issues related to services:
https://gravesoft.dev/in-place_repair_upgrade

n-tropy247, 2024-06-11:
I can confirm I am unable to mount esd files at all. Not sure what is going on with my system, but it goes beyond Rufus.

closed 2024-06-11 by n-tropy247