pbatard/rufus #2499: Windows 10 IoT Enterprise LTSC 2021 - Windows To Go - "Use the same regional options as this user's" failes to set time zone

JustMyGithub, 2024-06-11:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
As the log is huge an I am supposed to not redact it I did not provide it - feel free to close this because of this, but I wanted to let you know about this:.
Using "Use the same regional options as this user's" for Windows 10 iOT Enterprise LTSC 2021 (windows To Go) did correctly set the keyboard layout, but it failed to set the timezone correctly.
The iso name is en-us_windows_10_iot_enterprise_ltsc_2021_x64_dvd_257ad90f and I verified the checkums via heidoc.net
Rufus x64 v4.4.2103 (Portable)

Log
---
```
<FULL LOG>
```

pbatard, 2024-06-11:
I'm going to need the beginning and end of the log.

You can truncate a bunch of "Copying" messages, but I do need to see pretty much all of your log outside of this.

You can also zip the file and attach it to the issue. Or you can split it over multiple posts. There are more than one way to provide a log, even if GitHub says it's too large to copy/paste.

JustMyGithub, 2024-06-11:
Ok, so I replaced the GUIDs and the username, the Extracting-Creating-Lines and the Lines about all the drives that are detected.

As it might cause canfusion: I did a first run with another Win10 ISO that included the bad block check. During the check I noticed I picked the wrong but I waited for the bad block check to finish before I aborted. I removed that part from the log.

The checkdisk is in German, it says "no problems found, no actions required"

`Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.4412)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
[...]
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CES_X64FREO_EN-US_DV9'
  Detected: Windows 10 ISO (Build 19041.1288)
  Size: 4.5 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_10_iot_enterprise_ltsc_2021_x64_dvd_257ad90f.iso (4.5 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\%Username%\AppData\Local\Temp\RufF8D4.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\%Username%\AppData\Local\Temp\Ruf2573.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 IoT Enterprise LTSC' (Build: 19041, Index 2) for Windows To Go
Selected Windows User Experience options:
• Disable data collection
• Use '%Username%' for local account name
• Use the same regional options as this user's
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\%Username%\AppData\Local\Temp\Ruf7FAB.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 IoT Enterprise LTSC' (Build: 19041, Index 2) for Windows To Go
Selected Windows User Experience options:
• Disable data collection
• Use '%Username%' for local account name
• Use the same regional options as this user's
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\%Username%\AppData\Local\Temp\RufA19E.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 Enterprise LTSC' (Build: 19041, Index 1) for Windows To Go
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\%Username%\AppData\Local\Temp\RufAC1E.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 IoT Enterprise LTSC' (Build: 19041, Index 2) for Windows To Go
Selected Windows User Experience options:
• Disable data collection
• Use '%Username%' for local account name
[... 141 lines cut ...]
Partition type: GPT, NB Partitions: 3
Disk GUID: {%DISK-GUID%}
Max parts: 128, Start Offset: 17408, Usable = 2000398899712 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  Detected File System: FAT32
  ID: {%Part1-GUID%}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {%Part2-GUID%}}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {%Part3-GUID%}
  Size: 1.8 TB (1999991017472 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
`

pbatard, 2024-06-14:
Well, duplicating the regional options does not implicitly mean duplicating the timezone settings, and the explicit purpose of Rufus with this is to avoid the user being interrupted by the installation process to select its regional options. So, the fact that it applies a timezone that is decided by the default regional option and might not be the actual one the user wants is outside of the [actual design goals](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Could_you_add_support_for_removing_ltWindows_option_Xgt_in_the_Windows_User_Experience_dialog) of Rufus with this feature, which means that we don't really have a reason to want to address this (since again, our goal is to shortcircuit any element that make the Windows installer prompt the user instead of pressing forward with the installation, so that the user can get to a fully installed and running Windows, **that they can then finish customizing to their liking**).

But I hear you. It would be nice if the timezone could also be duplicated when the user selects to duplicate the regional option, so, considering that [there is an answer file setting that does just that](https://learn.microsoft.com/en-us/windows-hardware/customize/desktop/unattend/microsoft-windows-shell-setup-timezone), I'll see what I can do (though I have to state that, since Windows requires the use of crappy designators such as `Pacific Standard Time_dstoff`, if there isn't a Windows API that allows us to get these text designators from the current timezone, and instead we have to go the extra mile to figure them out, I may very well choose to drop this request altogether).

JustMyGithub, 2024-06-14:
Well, regional settings of the creating OS were Germany, German keyboard, timezone Germany, German language, German regional format - its no non-default combination like German keyboard and Australien timezone. The IoT ISO does not provide German language, of course, but keyboard was set correctly and so was the country and the regional format.

I do not expect anything and I am thankful for rufus (without it I might have to use shitty tools like balenaEtcher or even pay for an other tool), I just wanted to let you know because I thought this was clearly a bug. If timezone is not part of "regional options" I just suggest to mention this when the option is shown to the user (like "Use the same regional options as this user's (timezone not included)" or similar. I assume that many people expect that timezone is an "regional option"

You removed the "Windows 10 IoT Enterprise LTSC 2021" from the title, are you sure that this does apply to non-IoT ISOs as well? It's been a while since my last Win 10 Pro Windows To Go, but I do not remember that I had to set the timezone in Windows to Go back then - but I may be wrong about this. Or maybe each ISO has a default timezone and for the en IoT Enterprise that is a US timezone and for the German Pro ISO that is timezone Germany so maybe Rufus did not tranfer the setting but as the default timezone of the ISO happens to be the one of the creating host OS I did not notice that it did not do that.

pbatard, 2024-06-14:
> You removed the "Windows 10 IoT Enterprise LTSC 2021" from the title, are you sure that this does apply to non-IoT ISOs as well?

Yes, because if I set timezone for nonstandard region for non IoT (e.g. German region with Australian timezone), then regardless of whether this was an IoT issue (which I will **NEVER** look into) or not, this will either get fixed for IoT, or this is a pure Windows bug that will have nothing to do with Rufus in the first place.

All Rufus can do is set region/timezone settings from the answer file. If that doesn't work, then it means that it's a Windows issue with how it processes answer files (since I'm pretty confident Rufus does set it exactly as Microsoft documents) and not a Rufus issue. As such, the part about it not working for IoT or Windows To Go becomes irrelevant, and the only part I am planning to look into, since it's not something Rufus is doing right now, is carry the timezone settings from the user when they select to use the same regional options and, outside of trying to keep issue titles short (because a title shouldn't be the place to give specifics, but to briefly summarize the nature of the problem), that why I changed the title to something more digestible.

closed 2024-07-08 by pbatard

JustMyGithub, 2024-07-09:
Awesome just another time. Thanks!