pbatard/rufus #2500: missing: line 81: aclocal-1.14: command not found

Marietto2008, 2024-06-12:
Hello.

In relation to this fork :

https://github.com/markpeek/rufus/tree/markpeek-bhyve

that wants that Rufus detects an USB stick when it is passed as a bhyve virtual disk, I tried to compile that code,but I'm having a problem to compile it in my Ubuntu (23.10) installation :

```
root@Z390-AORUS-PRO-DEST:/home/ziomario/Scaricati/rufus# make

CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/ziomario/Scaricati/rufus/missing aclocal-1.14 
/home/ziomario/Scaricati/rufus/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [Makefile:277: aclocal.m4] Error 127

root@Z390-AORUS-PRO-DEST:/home/ziomario/Scaricati/rufus# apt reinstall aclocal-1.14
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package aclocal-1.14
E: Couldn't find any package by glob 'aclocal-1.14'

root@Z390-AORUS-PRO-DEST:/home/ziomario/Scaricati/rufus# apt search aclocal
Sorting... Done
Full Text Search... Done

root@Z390-AORUS-PRO-DEST:/home/ziomario/Scaricati/rufus# automake --version
automake (GNU automake) 1.16.5
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <https://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.
```

I prefer to compile it under Linux than under Windows,where I have 0 experience.

thecatontheceiling, 2024-06-12:
It's pretty funny that you expect Pete to help you compile **someone elses** fork of Rufus for some reason. Why not ask the person who made the fork instead?

pbatard, 2024-06-12:
> I prefer to compile it under Linux

And therein lies your problem. Rufus is a __Windows-only__ application that is designed to be compiled under a Windows environment (because it genuinely makes as little sense to try to support Linux compilation of a Windows application as it does to support compilation of, say, the Linux Kernel, under a Windows environment and using a Windows compiler like MSVC).

If you install MinGW on Linux, you **may** be able to compile it there, but I'm afraid I am not going to help you accomplish that.

Or you can just clone the repo, apply your (or somebody else's) changes and let the default [GitHub Actions](https://github.com/pbatard/rufus/actions) compile the binaries for you, as you will then get the Windows binaries in the produced GitHub Actions artifacts.

Marietto2008, 2024-06-12:
I will try to compile it in Windows by reading some documentation and asking all around. Maybe after having tried for some days,if I will find an error that I I will not have been able to fix,I will ask for some help...

Marietto2008, 2024-06-14:
I've been able to compile the code using visual studio 2022 and it worked. Very thanks.

pbatard, 2024-06-14:
OK. I'm going to close this issue then.