pbatard/rufus #2502: Unable to exclude requirement for Microsoft account from Windows 11 installer media

kovacsakos91, 2024-06-14:
Issue description
-----------------
I use Windows 11 multi edition ISO to create a bootable USB. Process is successful, pendrive is bootable. Windows installer is starting, files are copying to SSD. It seems successful, without any error message. After reboot I get error message on loading screen.
Type of notebook: Acer Aspire 3 A315-510P-36PG

Log
---
[image]

thecatontheceiling, 2024-06-14:
Thanks for pasting in a screenshot of a random error written in a language that the Rufus developer doesn't understand in the **Log** section while completely ignoring the instructions for creating a proper issue.

kovacsakos91, 2024-06-28:
Thanks a lot for your help, it is resolved. For future reference, this video contains solution: https://www.youtube.com/watch?v=wcsnzAtPNCI