pbatard/rufus #2503: windows user experience does not display

melonges, 2024-06-14:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
The image is official, downloaded from the Microsoft website
Log:
---
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Pro x64 (Build 22631.2861)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found UAS (USB 2.0) device 'Samsung Portable SSD T5 UAS Device' (04E8:61F5)
NOTE: This device is a USB 3.1 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1887, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {B608C4AD-691B-443C-9982-6406036D6036}
Max parts: 128, Start Offset: 17408, Usable = 15524133376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {0E0A4876-90AB-4510-A986-049D287345CB}
  Size: 14.5 GB (15522004992 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {AAB9C564-C5B7-4A3E-A04E-7E505A5CA8F1}
  Size: 1 MB (1064448 bytes)
  Start Sector: 30318498, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
  Could not access image: [0x0000000D] The data is invalid.
Could not acquire WIM index
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 6.3 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_23H2_English_x64v2.iso (6.3 GB)
efi_image_parse: Invalid optional header magic 0

```

pbatard, 2024-06-14:
> ```
>   Could not access image: [0x0000000D] The data is invalid.
> Could not acquire WIM index
> ```

Your image is most likely corrupted.

Please perform the following **EXCEEDINGLY RELEVANT STEP**, which you chose to ignore from the check list, and report your SHA-1:
> - [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

melonges, 2024-06-14:
> For reference, with a correct, non-corrupted `Win11_23H2_English_x64v2.iso` image, this is what you should see (including the proper SHA-1 for that image -- if your SHA-1 is different, then your `Win11_23H2_English_x64v2.iso` image **IS** 100% corrupted):
> 
> ```
> Scanning image...
> ISO analysis:
>   Image is a UDF image
> Disk image analysis:
>   Image does not have a Boot Marker
> ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
>   Detected: Windows 11 ISO (Build 22621.2861)
>   Size: 6.3 GB (Projected)
>   Has a >4GB file
>   Uses: EFI
>   Uses: Bootmgr (BIOS and UEFI)
>   Uses: Install.wim (version 0.13.1)
> Using image: Win11_23H2_English_x64v2.iso (6.3 GB)
> 
> Computing hash for 'D:\Incoming\Win11_23H2_English_x64v2.iso'...
>   MD5:    b036c5867361d3ca52db74b16bacd586
>   SHA1:   cd5e995387669b2ff2c58c23894900f1d238a103
>   SHA256: 36de5ecb7a0daa58dce68c03b9465a543ed0f5498aa8ae60ab45fb7c8c4ae402
> ```

I'm sorry, I really ignore to check the checksum, because with this image I was able to install windows

closed 2024-06-14 by melonges