pbatard/rufus #2506: Samsung uSD Card Reader USB Device (04E8:61ED) detected as HDD

acuteaura, 2024-06-15:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
This device should be detected as a microSD card reader.

Log
---
```
Rufus x64 v4.5.2180
Windows version: Windows 11 Pro for Workstations x64 (Build 22631.3737)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Samsung uSD Card Reader USB Device' (04E8:61ED)
Using 'autorun.inf' label for drive G: '256 GB'
Device eliminated because it was detected as a Hard Drive (score 10 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
```

pbatard, 2024-06-15:
Can you please enable enumeration debug with <kbd>Alt</kbd>-<kbd>.</kbd> and post the log you get then. It should be a lot more verbose and will contain the information I need to have to investigate.

acuteaura, 2024-06-15:
```
Rufus x64 v4.5.2180
Windows version: Windows 11 Pro for Workstations x64 (Build 22631.3737)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Samsung uSD Card Reader USB Device' (04E8:61ED)
Using 'autorun.inf' label for drive G: '256 GB'
Device eliminated because it was detected as a Hard Drive (score 10 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_05e3&pid_0608#6&2bea64b6&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[092]: USB\VID_2541&PID_9711\7&2AF08DC7&0&3
  Found ID[107]: USB\VID_058E&PID_3864\7&2AF08DC7&0&1
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub30#5&38f471a6&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[036]: USB\VID_05E3&PID_0608\6&2BEA64B6&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#root_hub30#5&19c782b&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[003]: USB\VID_8087&PID_0032\6&3878EF1F&0&5
  Found ID[119]: USB\VID_2F24&PID_0135\6&3878EF1F&0&4
  Found ID[034]: USB\VID_045E&PID_028E\76976680
  Found ID[233]: USB\VID_04E8&PID_61ED\MC113522009724
Processing Hub 4:
Processing Hub 5:
Processing IDs belonging to 'USBSTOR':
  USB\VID_04E8&PID_61ED\MC113522009724
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'Micron_2400_MTFDKBK512QFM' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSamsung_uSD_Card_Reader_0006'
  Matched with ID[233]: USB\VID_04E8&PID_61ED\MC113522009724
  Matched with Hub[3]: '\\?\usb#root_hub30#5&19c782b&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'Samsung uSD Card Reader USB Device' (04E8:61ED)
Using 'autorun.inf' label for drive G: '256 GB'
Device score: +5+5=+10 → Detected as HDD
Device eliminated because it was detected as a Hard Drive (score 10 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'WD_BLACK SN850X 4000GB' => Eliminated
0 devices found
```

pbatard, 2024-06-15:
Thanks. I'll see what I can do to have your device automatically listed in the next Rufus release.

thecatontheceiling, 2024-06-16:
Do you manually add devices to a whitelist if they're being incorrectly detected as an HDD?

pbatard, 2024-06-16:
It's not a whitelist per se, but we do filter out devices by assigning them a score depending on the type of the drive (FIXED vs REMOVABLE) it's size as well as various other factors listed in https://github.com/pbatard/rufus/blob/master/src/hdd_vs_ufd.h.

In this case, I suspect we're going to need to lower the score when we identify `SD Card` (with a space instead of a dash) as part of the device string.

weskerty, 2024-06-20:
Why not add the detection of Internal Disks through a key combination?
It would be useful to install Linux or WToGo from Windows itself.

acuteaura, 2024-06-20:
> Why not add the detection of Internal Disks through a key combination? It would be useful to install Linux or WToGo from Windows itself.

 https://github.com/pbatard/rufus/wiki/FAQ#do-you-plan-to-support-internal-hddsssds

thecatontheceiling, 2024-06-21:
> Why not add the detection of Internal Disks through a key combination?
> It would be useful to install Linux or WToGo from Windows itself.
> 

Since when is Rufus an operating system installer?

weskerty, 2024-06-21:
> Also, could you not hijack the issue?

Are the users the problem?
delete users

> Since when is Rufus an operating system installer?
You can now do it with Windows from a drop-down menu. WindowsToGo

pbatard, 2024-06-21:
> Are the users the problem?

This is not a department store. The customer is **not** always right.

You were directed to the FAQ entry that answered your question exactly. And you **did** piggyback on an issue to ask an unrelated question, which is very much frowned upon (because then you waste time for everybody, including yourself, by not following the check list that would have directed you to the FAQ and to the answer your sought).

> delete users

Trust me, I have zero issues banning users who think they are above common decency and think their questions or problems **entitles** them to hijack an existing issue. Now, if you do want to find it out the hard way, please continue adding unnecessary comments after your unrelated question has already been answered...

pbatard, 2024-07-08:
The issue reported by OP will be fixed in the next release of Rufus.

commit 83b1e730 (2024-07-08): [dev] add exception for Samsung uSD Card devices

* Closes #2506.
closed 2024-07-08 by pbatard