pbatard/rufus #2507: Rufus closes when you click on the start button with Windows  11 23h2

MauroSupera, 2024-06-15:
Hi everyone this  is just  a message for those had the same issue trying to create a bootable flash  just use another version of the  rufus for eg 3.21 it will work properly

JonnyTech, 2024-06-15:
﻿In order to help others, please follow the checklist and post a log so that the issue can be investigated and fixed:

PLEASE READ THIS CAREFULLY:

1. You *MUST* read and complete the steps from the checklist below, by placing
   an x into each [ ] (so that it shows '[x]', NOT '[ x]' or '[x ]'), BEFORE
   clicking on 'Submit new issue'.

2. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will
   usually result in your issue being dismissed without notice.

3. If you are reporting an issue when trying to run Rufus, or when trying to
   boot a media created by Rufus, you *MUST* provide a log, period. Please do
   not assume that the developer(s) will be able to "guess" the specifics of
   your environment, what image you used, what type of media you used it with
   or the many many other critical parameters that the log provides data for.
   To investigate an issue, a log from Rufus is ALWAYS required.

4. If you still *choose* not to provide a log when reporting a problem, you
   agree that your issue will be closed without any further investigation.

YOU HAVE BEEN WARNED.

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2024-06-15:
No log. Nothing I can do. As mentioned in the check list, even if Rufus closes, you can use DbugView to provide the log data.

closed 2024-06-15 by pbatard