pbatard/rufus #2509: Only ask

seba2282, 2024-06-20:
Hello, 
I don't remember good. 
When in Rufus i put options disable data collections (skip questions about privacy) and put regional what use user, after put that options to create pendrive by Rufus and indtall Windows 10/11 turn on localization? Or it will be blocked in windows settings how some options are about collections with write management by the organization? 

Best Regards

seba2282, 2024-06-20:
I met situation, I putted some option in Rufus, he made some policy and when was first configure and i saw desktop, settings took, then some buttons was inactive and wrote how i told in issue. And i only wanted know if seriously not put nothing in policy. Best regards :)

pbatard, 2024-06-20:
Rufus does not alter any policy for the installation of Windows. It only creates [an answer file](https://learn.microsoft.com/en-us/windows-hardware/manufacture/desktop/update-windows-settings-and-scripts-create-your-own-answer-file-sxs?view=windows-11). Therefore, whatever you are seeing on the platform where you installed Windows using a media you created with Rufus comes from Microsoft, and you will have to ask Microsoft about it.

closed 2024-06-20 by pbatard

seba2282, 2024-06-20:
Thanks for answer.