pbatard/rufus #2512: rufus cannot download windows 8.1.iso , i mean rufus error "could not retrieve languages from server"

zacky-I, 2024-06-29:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pineapple63, 2024-06-29:
Seems like Microsoft removed the downloads from their server

thecatontheceiling, 2024-06-29:
Yes, it's been removed for a very long time by now.

pbatard, 2024-07-08:
Thanks for the report. I have updated the Fido script in pbatard/Fido@29d8150b04fb8d1bd9e90a3e51834c05ab570dcd, and therefore the change will also apply to Rufus.

closed 2024-07-08 by pbatard