pbatard/rufus #2516: Rufus not recognize typical cheap USB Floppy Drive.

psycodeoday, 2024-07-05:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

It is a cheap typical USB Floppy Drive, which you can buy today on Amazon, or eBay, AliExpress.

After plug in, Windows recognize it, provide drivers automatically, and then it is ready to use as a drive A.

It read and write floppy disk without any issues

But Rufus doesn't see it. No matter which version I use, the drive selection window is empty.

In Device Manager Tree, I see Floppy Disk Drive,  and in Device properties I have
**Hardware IDs:**
USBSTOR\SFloppyTEAC____USB_UF000x______0.00

Please fix it. Looking for the amount of sold items I think that this is popular drive, and a lot of people use very old computers now for fun, or nostalgia, so the ability to create a bootable floppy disk in Rufus using USB floppy drive could be very helpful.

BTW.
3rd party programs:
Universal USB Installer: recognize is as drive A
(old) HP USB Disk Storage Format Tool: not see it as well.

Windows 10 64bit 22H2 10.0.19045.4529

screenshots:

[image]

[image]

[image]

[image]

[image] 

[image]

thecatontheceiling, 2024-07-05:
Please provide a log... come on man

pbatard, 2024-07-05:
Rufus has a deliberate hard limit of larger than 8 MB drives (meaning anything lower is eliminated by default). Also floppy and HDD boot are completely different beasts for BIOS (i.e. if you're hoping to use a flat disk image to boot using a floppy under BIOS using Rufus, even if Rufus let you do so, it wouldn't work).

There are also hard limits for NTFS formatting that make it unsuitable for floppy usage.

In other words, if I were to try to add floppy support in Rufus, I'd have to add a lot of extra code for next to no benefits. That's not going to happen.

If you want software that works with floppy drives, I will respectfully ask that you look elsewhere.

closed 2024-07-05 by pbatard