pbatard/rufus #2517: Feature Rquest: Plain image for OEM preinstalled laptops

Raart, 2024-07-08:
With the advent of SSDs and the quick evolution of their characteristics/capacity it is often interest of the owner of a laptop to have more disk space. At the same time installing Windows from the downloaded ISOs doesn't guarantee the full functionalities of the laptop cause missing of OEM drivers, apps (and bloatware) being stored in the hidden partition on the SSD. Mostly the installed Windows version differs from the standard one due to OEM adaptations to the specific hardware.
In case of a full reset (e.g. in case of sell of the laptop or serious crash recoveries), parts or the entire installation Windows files are taken from another hidden partition. It has to be noted that to get the full reset process to end with success, the physical position of this partitions is important as they are "hardcoded" in the recovery processes.
It would be very interesting to have RUFUS being capable to make a .ISO image that contains all this partitions and procedures to make it possible to grant the various recover/resume/full reset functions of Windows on laptops. It would be of great help in case of severe crash recoveries caused by SSD crash or partitions damages, or in installations on external USB SSDs etc..

pbatard, 2024-07-08:
> It would be very interesting to have RUFUS being capable to make a .ISO image that contains all this partitions and procedures

That's too custom a request which would involve an awful lot of investment from the developer (how am I supposed to test and maintain this without getting my hands on a whole slew of laptops from a whole range of manufacturers, as well as invest **A LOT OF EXTRA TIME** ensuring that Rufus can properly access all of this SUPER CUSTOM PROPRIETARY data) with little benefits.

I'm sorry but, because I am providing a _free_ application that is tangentially related to what you want to accomplish, you cannot ask me to supersede what **your** laptop manufacturer is supposed to do, which is to provide you with the means of reinstalling your laptop from scratch, even in the case of a catastrophic hardware failure.

That goes way beyond the scope of what I am designing Rufus for, as such, I am not planning to ever implement your request. See also https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.

Closing as not planned.

closed 2024-07-08 by pbatard

Raart, 2024-07-08:
Thanks for the prompt answer. I respect and understand your position and don't insist on my request even if I think that this largely involves an enormous and fast growing number of laptops compared to desktops. BTW - Thanks for your valuable work on Rufus I use extensively as I'm not a one-operating-system guy. Regards, Andreas