pbatard/rufus #2519: oobeSystem is gone?

prlw1, 2024-07-09:
1. I buiilt Rufus from today's head: https://github.com/pbatard/rufus/commit/10d33c66588bd31070ddc3b45deb1cd83c0aeae0
2.  I used the microsoft media creator to download today's windows 11 disk image file (Build 22621.22861)
3. I generated the USB stick with options:
- Remove requirement for an online Microsoft account
- Create a local account with username xxx
- Set regional options to the same values as this user's
- Disable data collection (Skip privacy questions)
- Disable BitLocker automatic device encryption
5. After a successful boot, "getting ready" and another boot, proceedings halt with

> Windows could not parse or process unattended answer file [C:\Windows\Panther\unattend.xml] for pass [oobeSystem]. A component or setting specified in the answer file does not exist

"OK"

> Windows could not complete the installation. To install Windows on this computer, restart the installation.

My guess is that `oobeSystem` to `Remove requirement for an online Microsoft account` is gone?

(A pain as my only account is provided by work, and the installer rejects work accounts - why?!)

pbatard, 2024-07-09:
Log, Log, Log, LOG!!!!

You are asked for a log. It is not up to you to decide whether it's needed or not.

I simply cannot answer your question as you are not providing enough data about the **actual** environment Rufus ran in. You believe you do, but you actually don't.

Also, `oobeSystem` is unrelated to local account creation. Instead it deals with the BitLocker and data collection option.

Since you are taking a shortcut, and did not provide a log, I will take one too: My guess is that your issue has everything to do with BitLocker, so please disable the BitLocker option and report what happens.

prlw1, 2024-07-09:
I rewrote the USB stick to get the log, @pineapple63  has posted an equivalent - thanks!

I tried again just deselecting the bitlocker option and saw the same apparent behaviour.

I failed to boot another OS to get at the `Panther` directory.

pineapple63, 2024-07-09:
From what i can tell, it seems to be the regional options selection that causes the error
I created a USB using a retail ISO image to see if something funky was going on with the media creation tool ISOs
[rufus3.log](https://github.com/user-attachments/files/16151177/rufus3.log)
and the USB created from the retail ISO gave the same error

pbatard, 2024-07-09:
Then this has to be the new `TimeZone` setting from 10d33c66588bd31070ddc3b45deb1cd83c0aeae0.

I guess this needs to be moved out of `Microsoft-Windows-International-Core` and into `Microsoft-Windows-Shell-Setup`.

prlw1, 2024-07-10:
Indeed! Using rufus built from https://github.com/pbatard/rufus/commit/83b1e7306242884416beac23e0d977725897e217 got me a successful install!

pbatard, 2024-07-15:
This issue should be fixed in latest builds.

closed 2024-07-15 by pbatard

realsetvin, 2024-09-27:
> This issue should be fixed in latest builds.

Issue still persists on Windows Home S

pbatard, 2024-09-27:
S is restricted by Microsoft itself. Nothing Rufus can do about that.

If you don't want the restrictions of running Windows in S Mode, see [here]
(https://support.microsoft.com/en-us/windows/windows-10-and-windows-11-in-s-mode-faq-851057d6-1ee9-b9e5-c30b-93baebeebc85).

And for the record, S Mode is persisted **even if you go through a full system wipe and reinstall**.

commit 78608c35 (2024-07-15): [wue] fix TimeZone option not being applied to the right parent

* Closes #2519.
* Also update MinGW's base WINVER to Windows 10.