pbatard/rufus #2529: Rufus Windows To Go Not Installing

vince897, 2024-07-26:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [xx ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

-----------------
[image]
[Rufus x64 v4Report.docx](https://github.com/user-attachments/files/16395217/Rufus.x64.v4Report.docx)
<Please describe your issue here>
I tried several copies of Rufus 4.5p and several copies of Windows 11 ISO andRufus failed to complete to obtain a Windows To Go file.
To determine if my ISO file and Rufus were error free I ran Rufus for the Standard Windows Installation and it completed. I assume this verifies my ISO file and the version of Rufus 4.5p I am using.

The full Log is attached

Log
---
```
<FULL LOG>
```

pbatard, 2024-07-26:
1. Please don't copy paste the log into a Word document. It's absolutely dreadful to try to work with it, and I had to update your post to make it a lot more palatable. You can just zip the log file and attach that if it's too long.
2. ` Could not apply image: [0x00000570] The file or directory is corrupted and unreadable.` makes it pretty clear that the `install.esd` from your ISO image (which appears to be a **custom** Windows ISO, as any official ISO, such as the one that you can download through Rufus, would use an `install.wim`) is corrupted.
3. If the content of a `.esd` is corrupted, Rufus will have no trouble creating an installer, since it copies the `.esd` wholesale. Contrary to what you assume, Rufus does **not** validate that a `.esd` or `.wim` contains proper data when it creates a standard Windows installation media. In fact, the file could be all zeroes or complete garbage, and Rufus would not show any error. On the other hand, when creating a Windows To Go media, it needs to actually extract the content from the `.esd` so if that `.esd` is corrupted, you will see an error there, but not on regular media creation.

If you want to create a Windows To Go media, please make sure that you work with an official retail Windows ISO, such as the ones that Rufus can download (see below) and in case of any issue, make sure you [validate the SHA-1 of that ISO](https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_can_I_validate_that_a_Windows_ISO_is_a_genuine_retail_version) (which you won't be able to do with an unofficial or MCT ISO).
[image]

As it appears to be a corrupted ISO problem, I will close this issue. But feel free to open a new one if you still see a problem when using an **official retail** ISO.

closed 2024-07-26 by pbatard

vince897, 2024-07-26:
I'm sorry if I caused you a problem with the info that I sent.

I have used at least four downloads of Windows 11 all from Microsoft some using the Media Creation Tool. I don't know what a custom ISO file is.

I'll try to figue out how to do the the SHA test and download the Windows 11 ISO again.
Appreciate your time