pbatard/rufus #2530: Windows To Go ISO Extraction Failure

vince897, 2024-07-28:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have a computer with 32GB of soldered menory that is too small to update Windows and install other Programs.
Would llike to put Win11 To Go on a USB drive
I have used ISO files from Microsoft using Media Creation Tool and from Rufus confirmed by SHA256 agreement with multiple failures.
I am able to download the Standard Windows Installation  on the USB drive that works.
Log is for Rufus ISO file confirmed by SHA256 that fails with messag: Error Image Extraction Failure.
Would appreciate any help you can give.

Log
---
[rufus.log](https://github.com/user-attachments/files/16404161/rufus.log)

```
<FULL LOG>
```

pbatard, 2024-07-28:
First of all:

> - [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Oh really? Where's the SHA1 from your log then?

**DON'T** just tick the checkbox because it annoys you to leave them empty. Perform what's being requested of you and **then** check the box. Otherwise, since you did **NOT** produce an SHA-1 from your image that we can validate, I don't see why I shouldn't just assume that your ISO is corrupted and close this issue.

The checkboxes are not just there to look pretty. They are there because they are associated with a very valuable check, that you should perform if you want people to take you seriously...

Anyway:

> ```
> WARNING: The following application(s) or service(s) are accessing the drive:
> ● [4832] C:\Windows\System32\svchost.exe -k LocalSystemNetworkRestricted -p -s StorSvc (rwx)
> You should close these applications before retrying the operation.
> ```

along with:

> ```
> ● Creating EFI System Partition (offset: 1048576, size: 260 MB)
> Write error [0x00000005] Access is denied.
> Retrying in 5 seconds...
> ```

and:

> ```
> WIM processing retry: D:\Program Files (x86)\Microsoft\EdgeWebView\Application\100.0.1185.36\msedge.dll [[0x00000015] The device is not ready.]
> WIM processing error: D:\Windows\WinSxS\amd64_microsoft-windows-pnpsysprep_31bf3856ad364e35_10.0.22621.2506_none_2197e11432b1e729\sppnp.dll [[0x00000015] The device is not ready.]
> WIM processing error: D:\Program Files (x86)\Microsoft\EdgeWebView\Application\100.0.1185.36\msedge.dll [[0x00000015] The device is not ready.]
>  Could not apply image: [0x00000015] The device is not ready.
> ```

Makes it pretty clear that you have a process or application external to Rufus that is interfering with your drive.

You will need to fix that, because, obviously, Rufus cannot do much of anything if an external application is interfering with the media creation.

See also https://github.com/pbatard/rufus/wiki/FAQ#user-content-Error_0x00000005_Access_is_denied.

vince897, 2024-07-28:
Sir:

I  appreciate the help and will try to determine the conflict.

Not sure why you assumed I did not carry out the checksum for the ISO file,
I attach the test as showm below and the code agrees.
[image: image.png]

Vincent L. Pisacane, PhD
32304 Ann's Choice Way
Warminster, PA 18974-3378
Phone: 215-323-4628
Email: ***@***.***
Call Sign: WB3IDW

pineapple63, 2024-07-28:
Images and Files attached by email are discarded by github
If you want to post an image, either upload it to a third party site and link to it, or go to github and post the image directly to the issue tracker
https://github.com/pbatard/rufus/issues/2530

pbatard, 2024-07-28:
> Not sure why you assumed I did not carry out the checksum for the ISO file

I'm not assuming anything. Your log is missing the relevant entry if you had done it, which would look something like this:

```
Computing hash for 'D:\Incoming\!Win11_23H2_English_x64v2.iso'...
  MD5:    b036c5867361d3ca52db74b16bacd586
  SHA1:   cd5e995387669b2ff2c58c23894900f1d238a103
  SHA256: 36de5ecb7a0daa58dce68c03b9465a543ed0f5498aa8ae60ab45fb7c8c4ae402
```

Your log has no such section.

pbatard, 2024-08-10:
No update &rarr; Closed.

closed 2024-08-10 by pbatard