pbatard/rufus #2531: Rufus Crash in Windows 10

MominIqbal-1234, 2024-07-29:
[image]

both version not working 
click start button rufs  Crash

TCFFan123, 2024-07-30:
Attach the log

pbatard, 2024-07-30:
Yes, the checklist, which you ignored, and which explicitly requests a log, documented how you can produce a log even if the application crash using DebugView. And at any rate, you could copy the log from Rufus itself before you press START.

I'm afraid I have *no* time for people who cannot follow a simple checklist, do not even provide the name of the ISO or Boot selection mode they were using, and wrongly assume that, if they experience an issue, then everybody will be able to replicate it.

As a result, and as you were explicitly warned when you opened your request, I will close this issue.

closed 2024-07-30 by pbatard