pbatard/rufus #2538: Add support for compiling in newer msys2 env

ArchGuyWu, 2024-08-13:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
In ucrt64:
```
  In file included from C:/_/B/src/rufus/src/libcdio/iso9660/iso9660_private.h:29,
                   from C:/_/B/src/rufus/src/libcdio/iso9660/iso9660.c:28:
  C:/_/B/src/rufus/src/libcdio/config.h:43:18: error: static declaration of '_fseeki64' follows non-static declaration
     43 | #define fseeko64 _fseeki64
        |                  ^~~~~~~~~
  In file included from C:/_/B/src/rufus/src/libcdio/iso9660/iso9660.c:42:
  D:/M/msys64/ucrt64/include/stdio.h:650:23: note: previous declaration of '_fseeki64' with type 'int(FILE *, long long int,  int)' {aka 'int(struct _iobuf *, long long int,  int)'}
    650 |   _CRTIMP int __cdecl _fseeki64(FILE *_File,__int64 _Offset,int _Origin);
        |                       ^~~~~~~~~
  make[2]: *** [Makefile:307: libiso9660_a-iso9660.o] Error 1
  make[2]: *** Waiting for unfinished jobs....
  make[2]: *** [Makefile:313: libiso9660_a-iso9660_fs.o] Error 1
  make[2]: Leaving directory '/C/_/B/src/build-UCRT64/src/libcdio/iso9660'
  make[1]: *** [Makefile:521: all-recursive] Error 1
  make[1]: Leaving directory '/C/_/B/src/build-UCRT64/src'
  make: *** [Makefile:287: all-recursive] Error 1
```

In clang64:
```
  Making all in src
  make[1]: Entering directory '/C/_/B/src/build-CLANG64/src'
  Making all in ../.mingw
  make[2]: Entering directory '/C/_/B/src/build-CLANG64/.mingw'
    SED      ../../rufus/.mingw/dwmapi.def
    SED      ../../rufus/.mingw/version.def
    SED      ../../rufus/.mingw/wintrust.def
    LIB      dwmapi-delaylib.lib
    LIB      version-delaylib.lib
    LIB      wintrust-delaylib.lib
  OVERVIEW: llvm-dlltool
  
  USAGE: llvm-dlltool [options] file...
  
  OPTIONS:
    -D <value>              Specify the input DLL Name
    -d <value>              Input .def File
    -f <value>              Assembler Flags
    -k                      Kill @n Symbol from export
    -l <value>              Generate an import lib
    -m <value>              Set target machine
    --no-leading-underscore Don't add leading underscores on symbols
    -S <value>              Assembler
    -t <value>              Prefix for temporary files (ignored)
  
  TARGETS: i386, i386:x86-64, arm, arm64
  OVERVIEW: llvm-dlltool
  
  USAGE: llvm-dlltool [options] file...
  
  OPTIONS:
    -D <value>              Specify the input DLL Name
    -d <value>              Input .def File
    -f <value>              Assembler Flags
    -k                      Kill @n Symbol from export
    -l <value>              Generate an import lib
    -m <value>              Set target machine
    --no-leading-underscore Don't add leading underscores on symbols
    -S <value>              Assembler
    -t <value>              Prefix for temporary files (ignored)
  
  TARGETS: i386, i386:x86-64, arm, arm64
  make[2]: *** [Makefile:376: dwmapi-delaylib.lib] Error 1
  make[2]: *** Waiting for unfinished jobs....
  make[2]: *** [Makefile:376: version-delaylib.lib] Error 1
```

In mingw64, it compiled successfully, but didn't install its executable properly:
```
  [MSYS2 CI] mingw-w64-rufus: File listing diff for mingw-w64-x86_64-rufus
  
  warning: database file for 'ci' does not exist (use '-Fy' to download)
  error: package 'mingw64/mingw-w64-x86_64-rufus' was not found
  --- /dev/fd/63	2024-08-12 14:46:36.000000000 +0000
  +++ /dev/fd/62	2024-08-12 14:46:36.000000000 +0000
  @@ -0,0 +1,5 @@
  +/mingw64/
  +/mingw64/share/
  +/mingw64/share/licenses/
  +/mingw64/share/licenses/rufus/
  +/mingw64/share/licenses/rufus/LICENSE
```

ognevny, 2024-08-13:
> ```
>   [MSYS2 CI] mingw-w64-rufus: File listing diff for mingw-w64-x86_64-rufus
>
>   warning: database file for 'ci' does not exist (use '-Fy' to download)
>   error: package [...]

obviously install target is missing and it should be installed manually

pbatard, 2024-08-13:
1. Clang is **not** supported.
2. `make install` is **not** supported.
3. I am not seeing any of these errors in the environments I am running (and I am running latest from msys2)

Basically, unless I am seeing a compilation issue when invoking `configure`/`make` on an up to date MSYS2/MinGW systems **installed on Windows** or during GitHub Actions compilation, I am not going to do anything about it.

I *may* look at the patch, if you want to send one my way, but I am not going to investigate stuff that I am not seeing breaking on my side (or when using a different compiler than `gcc`), and I am not seeing any of the issues you mention.

&rarr; Closing

closed 2024-08-13 by pbatard

ognevny, 2024-08-13:
there are many environments in msys2: https://www.msys2.org/docs/environments/

compilation fails for ucrt64, but not mingw64 (ucrt also uses gcc as main C compiler)

pbatard, 2024-08-13:
Well, I'm afraid that, unless ucrt64 is installed as the default env by msys2 (which it doesn't appear to be since neither my home msys2 or the msys2 toolchain from GitHub Actions have an issue), I am not planning to support it.

As I said however, if you want to send a patch to fix ucrt64 compilation, I will try to look at it (but I will not promise to apply it, or maitain uct64 compilation in working order going forward *if* I apply it).

In short, if you want to use something else that the default environments and toolchains, you are on your own as I genuinely don't have time to spare to try to check them and make them work.

ArchGuyWu, 2024-08-13:
> Well, I'm afraid that, unless ucrt64 is installed as the default env by msys2 (which it doesn't appear to be since neither my home msys2 or the msys2 toolchain from GitHub Actions have an issue), I [...]

In the above link, it says: If you are unsure, go with UCRT64.
And if you install msys2 by using msys2-x86_64-*.exe, it will automatically launch ucrt64 after installation, as it says: Now MSYS2 is ready for you and a terminal for the [UCRT64 environment](https://www.msys2.org/docs/environments/) will launch.

pbatard, 2024-08-13:
Except we are compiling for MinGW, not msys2.

That msys2, because it supports multiple toolchain, recommends one that is **not** that is referenced all over the place in our readme and website is irrelevant. They could recommend a super obscure one for what is worth, but as long as they support MinGW, and we explicitly state:

> Use either Visual Studio 2022 or MinGW

under *Compilation* in the readme, you are **obviously** meant to install MinGW and **not** ucrt when using msys2.

ognevny, 2024-08-13:
> under *Compilation* in the readme, you are **obviously** meant to install MinGW and **not** ucrt when using msys2.

the only difference between UCRT64 and MINGW64 is C runtime. UCRT still uses MinGW, but it's based at UCRT, not MSVCRT