pbatard/rufus #2541: [Bug] 官方镜像通过Rufus刷写后，提示“由你的组织管理”

ukhack, 2024-08-14:
我使用rufus将最新的23h2刷入优盘，并且开启了类似跳过最低内存检查等一些设置。
插入优盘更新了我的系统，最终系统变成了“由你的组织管理”
[image]

[image]

很多设置都被ban掉了，类似如通过防火墙的按钮都无法点击确认
Many settings are baned

thecatontheceiling, 2024-08-14:
Rufus doesn't set any policies in the ISO file, even if you use the User Experience check boxes.

https://github.com/pbatard/rufus/blob/20881ceea6f4e941dc4441a448866fb8292f5ef8/src/wue.c

pbatard, 2024-08-14:
Please be mindful that the **automatic** license activation, that is applied by Microsoft and that is tied to your hardware, ultimately controls what features of Windows are allowed or not.

For instance, if your machine was ever activated for a Windows S Mode license, **Microsoft** will drastically restrict what you can do with Windows, and, again, this is 100% controlled by the type of license that was installed for your machine and **not** how you installed Windows.

At any rate I can guarantee that Rufus had nothing to do with your issue (and you didn't produce a log anyway) so I will close this issue.

closed 2024-08-14 by pbatard