pbatard/rufus #2542: RAMMUS 126 recovery image too large to copy to FAT32 disk.

gjlamb, 2024-08-15:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Latest recovery file for RAMMUS to create Chrome/Sebanc bootable disk is too large for FAT32.  Please add exFAT format option.

Log
---
```
<FULL LOG>
```

gjlamb, 2024-08-15:
It wasn't obvious at first but after researching exFAT for running Linux is a no go I thought i'd look at if NTFS works.  Who knew Linux would run on NTFS.  Tried NTFS format for imaging usb drive and copied files on successfully.  Drive booted and am doing install.  I'll update comment on instruction site to update that NTFS option is needed.

closed 2024-08-15 by gjlamb

pbatard, 2024-08-15:
At any rate, the exFAT formatting option you originally requested is **already** available from Rufus, and has been since the beginning.

Also, since this is a bootable recovery image, you wouldn't really copy it as it onto a USB drive, but instead write it in DD mode, which Rufus will do for you if you select the `.bin` or `.zip`, and where the filesystem becomes completely irrelevant.

I'm not sure what you *thought* your original problem was, but the request you made was unwarranted because Rufus **does** support exFAT formatting, and you should not need to format your drive to any file system in Rufus to use the rammus recovery image...

gjlamb, 2024-08-15:
I downloaded the latest version and it only had 2 formatting options…. FAT32 and NTFS. No other options. The default mode for writing is not DD mode.

pbatard, 2024-08-15:
Well, I too downloaded the latest version from https://chromiumdash.appspot.com/serving-builds?deviceCategory=Chrome%20OS (`chromeos_15886.44.0_rammus_recovery_stable-channel_mp-v5.bin.zip`) and, after selecting the `.zip`, I found that Rufus **properly** detected it as a compressed bootable image, and wrote it in DD mode.

And because it was detected as bootable image, the file system field was **greyed out** (which means that it is not applicable, as any greyed out field indicated), thereby indicating that that you could not select a file system (since, again, it is **iirelevant** for a DD image, and whatever is greyed out doesn't matter).

Of course, since you **chose** not to provide a log, despite being explicitly asked to do so when creating the issue, I cannot vouch for what you saw, but, again, because you **chose** not to provide a log, I can't help but express a lot of doubts about your report that you were able to switch beween FAT32 or NTFS after selecting the image.

By the way, here is the kind of log you were supposed to provide, which demonstrates that `chromeos_15886.44.0_rammus_recovery_stable-channel_mp-v5.bin.zip` was properly detected as a bootable disk image, and that Rufus did properly write it in DD mode, with no option to select a file system:

```
[Thu Aug 15 11:11:14 2024]
Rufus x64 v4.5.2180
Windows version: Windows 11 Pro for Workstations x64 (Build 22631.4037)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 0 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 12
Disk GUID: {93C920E7-6334-DB47-9A57-CF037B6CCAFB}
Max parts: 128, Start Offset: 17408, Usable = 31406914048 bytes
Partition 1:
  Type: Linux Data Partition
  Name: 'STATE'
  Detected File System: ext2
  ID: {D13BC833-BC88-2D4B-AD08-28B254353FAF}
  Size: 3.3 MB (3461632 bytes)
  Start Sector: 8773632, Attributes: 0x0000000000000000
Partition 2:
  Type: Chrome OS Kernel Partition
  Name: 'KERN-A'
  Detected File System: (Unrecognized)
  ID: {4F6A89E6-0A4C-B345-ADF6-D6D4068899E2}
  Size: 32 MB (33554432 bytes)
  Start Sector: 20480, Attributes: 0x01FF000000000000
Partition 3:
  Type: Chrome OS Root Partition
  Name: 'ROOT-A'
  Detected File System: ext2
  ID: {C9A677D2-C59B-254C-B381-75851759C333}
  Size: 4 GB (4294967296 bytes)
  Start Sector: 385024, Attributes: 0x0000000000000000
Partition 4:
  Type: Chrome OS Kernel Partition
  Name: 'KERN-B'
  Detected File System: (Unrecognized)
  ID: {FB1BA31C-CEDB-3E4C-B2D1-EBE5FCA3089E}
  Size: 32 MB (33554432 bytes)
  Start Sector: 86016, Attributes: 0x0000000000000000
Partition 5:
  Type: Chrome OS Root Partition
  Name: 'ROOT-B'
  Detected File System: (Unrecognized)
  ID: {00C2665D-9453-7446-8DFB-415E301502DF}
  Size: 2 MB (2097152 bytes)
  Start Sector: 380928, Attributes: 0x0000000000000000
Partition 6:
  Type: Chrome OS Kernel Partition
  Name: 'KERN-C'
  Detected File System: (Unrecognized)
  ID: {F464A340-01C7-6D42-A5AE-E0A0C53BA7A8}
  Size: 512 bytes (512 bytes)
  Start Sector: 16448, Attributes: 0x0000000000000000
Partition 7:
[... 128 lines cut ...]
  Start Sector: 16450, Attributes: 0x0000000000000000
Partition 10:
  Type: Chrome OS Reserved Partition
  Name: 'reserved'
  Detected File System: (Unrecognized)
  ID: {0F6B24B8-5844-E147-964C-757CD7937A62}
  Size: 512 bytes (512 bytes)
  Start Sector: 16451, Attributes: 0x0000000000000000
Partition 11:
  Type: {CAB6E88E-ABF3-4102-A07A-D4BB9BE3C1D3}
  Name: 'RWFW'
  Detected File System: (Unrecognized)
  ID: {17232508-EFCD-F04A-91EB-E151B7D07D3A}
  Size: 8 MB (8388608 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 12:
  Type: EFI System Partition
  Name: 'EFI-SYSTEM'
  Detected File System: FAT16
  ID: {800B6B2E-B4BF-AB4E-B103-BD53D97829C0}
  Size: 32 MB (33554432 bytes)
  Start Sector: 315392, Attributes: 0x0000000000000004
```

Now, if you still want to claim that Rufus somehow allowed you to select a file system, and that only FAT32 and NTFS were available when you did so, then you'd better produce a log of how exactly you proceeded, because, otherwise, I will continue to have to dismiss your claim about Rufus allowing you to select a file system when presented with a Rammus recovery image, and exFAT not being available from file system dropdown, as hearsay.

gjlamb, 2024-08-15:
I’m following instructions for building a Linux Mint bootable drive to
install Sebanc/Brunch install of chrome OS. I’m not imaging the RAMMUS
image with Rufus, I’m imaging Linux mint lite then copying the Chrome
recovery file to the boot drive to run the install script.

https://beebom.com/how-install-chrome-os-on-pc/amp/
<https://beebom.com/how-install-chro>

gjlamb, 2024-08-15:
I didn’t supply a log as I didn’t think it would be of any use as it wasn’t
an issue with the imaging process it was that I thought that following the
instructions I had that something had changed with the size of the recovery
file and that FAT32 didn’t work anymore.

Anyhow, I closed the issue and put in some info, if I could have deleted
the issue I would have done that. Feel free to not let it take up any more
of your time.

pbatard, 2024-08-15:
> I’m imaging Linux mint lite then copying the Chrome recovery file to the boot drive to run the install script.

And we were supposed to **GUESS** that how?!?!?

Now, of course **again**, if you had provided a log, **AS REQUESTED**, one might have had a clue that you were actually creating a Linux Mint boot disk, but the only data you mentioned was Rammus recovery, and there was not a single mention if Mint.

> I didn’t supply a log as I didn’t think it would be of any use 

Well, you were 100% wrong.

I'm not asking for a log to annoy people. I am asking for a log precisely because people erroneously think that what they are doing will be super obvious, and consistently fail to mention **critical** details, such as the fact that they are using Rufus to create a Mint bootable drive rather than using a ChromeOS recovery image in the way it is officially intended to be used.

So, of course, if you create a Mint bootable drive, then the only file systems available will be the ones that are supported by Mint, which means FAT32 or NTFS (as exFAT is usually less widely supported than NTFS, and I don't know of any recent distro that might support both, but where exFAT would present any advantage over NTFS).