pbatard/rufus #2543: rufus.ie riddled with ads and malware links

bonominijl, 2024-08-15:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2024-08-15:
If you see deceptive ads, please report them to Google [here](https://support.google.com/google-ads/answer/7660847?hl=en).

As much as I would like to be able to say I have control over the ads that are displayed on the Rufus webpage, and I am trying to filter the crappy ones (Gambling, Esoteric, Medical) out **as much as Google will let me**, the fact is that, ultimately, **Google** are the ones who control what ads display on rufus.ie.

So if you do see deceptive ads there, then you want to report them to Google, because I genuinely have no control whatsoever over what displays into the ad banners.

As to "malware links", if, as you say, rufus.ie is riddled with them, the you should have no trouble pointing to the links, in the page content (not the ads), that, according to you, point to malware.

So can you do that please?

pbatard, 2024-08-16:
No update &rarr; Closed.

closed 2024-08-16 by pbatard