pbatard/rufus #2544: Add a "Is it legal ?" question in FAQ

Mitia99, 2024-08-16:
Hi,

I'm aware that Rufus is legal but some new folks may wonder if it is.

A "**Is it legal ?**" question in FAQ will be great

Thank you,

pbatard, 2024-08-16:
Not gonna do that, because it will lead to people to think that Rufus might be operating in a grey area, whereas it is **not** doing so at all.

If you need to specify whether a product is legal, you are actually bringing Fear, Uncertainty and Doubt (FUD) about that product. And if some people *think* that Rufus might be operating in the verge of legality, whereas we are super transparent about everything we are doing and, as opposed to other products (who may embed files of dubious provenance) are taking great care to never ever even remotely run afoul of copyright law (which is the reason why, for a time, we removed the ability to create MS-DOS disks, on account that we hadn't devised how to obtain the MS-DOS files in a manner that would satisfy copyright laws), it's really those people's problems, because someone who is actually operating in a 100% legal framework should not have to justify themselves to people who **erroneously** assume they might not be (and don't want to educate themselves, on their own, on the fact that they actually are).

&rarr; Rejected.

closed 2024-08-16 by pbatard

thecatontheceiling, 2024-08-16:
What? In what jurisdiction is copying files to an USB illegal?

Mitia99, 2024-08-16:
> What? In what jurisdiction is copying files to an USB illegal?

I know that RUFUS isn't illegal but some folks asked me if tweaking a Microsoft ISO is thus my suggestion to add this in FAQ

JonnyTech, 2024-08-16:
> What? In what jurisdiction is copying files to an USB illegal?

It is not copying them to a USB that it "illegal" but rather the act of obtaining the files. Pete has gone to great effort to ensure that any files obtained by the Rufus download capability are sourced from verified sources, ie directly from Microsoft public servers for Windows / MSDOS, with a clear licence for usage.

pbatard, 2024-08-16:
> but some folks asked me if tweaking a Microsoft ISO is

Well, the thing that these folks don't appear to have researched is that the "tweaking" is simply adding options that **Microsoft themselves** have made available. So it's like asking if running a commandline utility with a specific flag is legal. If the flag was added by its developer, you can draw your own conclusions...